loboris/MicroPython_ESP32_psRAM_LoBo #3: xtensa tool chain appears to be 32-bit on 64-bit system

slzatz, 2018-01-05:
I certainly may be doing something wrong but when I cloned the new directory, I appear to get the 32-bit xtensa tool chain and not the 64-bit tool chain although I am running on 64-bit Ubuntu 17.10.  I believe this is the reason that I am getting build errors, which were initially `file not found` but after installing 32-bit architecture components the `file not found` goes away but still getting build errors.

When I looked into the deactivated repository, the xtensa tool chain was 64-bit.

slzatz, 2018-01-05:
Note that when I also added `libz.so.1` with

    sudo apt-get install zlib1g:i386[/code

compilation completed successfully but still seems better to have 64-bit tool chain v. having to add support for 32-bit to a 64-bit system.

closed 2018-01-05 by slzatz

loboris, 2018-01-10:
Fixed.