loboris/MicroPython_ESP32_psRAM_LoBo #5: Modules not presents

hargathor, 2018-01-06:
Hi and first thanks for this for of micropython that is really impressive !
I've made an update this morning from the version 2.0.8 to the versoin 3.1.0 and some of the modules I used were not present. ds18x20 or onwire for exemple.
Here is the output of help('modules') for official micropython 1.9.3
````
MicroPython v1.9.3-238-g42c4dd09 on 2018-01-02; ESP32 module with ESP32
Type "help()" for more information.
>>> help('modules')
__main__          framebuf          re                upip
_boot             gc                select            upip_utarfile
_onewire          hashlib           socket            upysh
_thread           heapq             ssl               urandom
apa106            inisetup          struct            ure
array             io                sys               urequests
binascii          json              time              uselect
btree             machine           ubinascii         usocket
builtins          math              ucollections      ussl
cmath             micropython       uctypes           ustruct
collections       neopixel          uerrno            utime
dht               network           uhashlib          utimeq
ds18x20           ntptime           uheapq            uzlib
errno             onewire           uio               websocket
esp               os                ujson             zlib
flashbdev         random            uos
Plus any modules on the filesystem
>>>
````
Here is the output of help('modules') for Loboris_Micropython 2.0.8
````
MicroPython ESP32_LoBo_v2.0.8 - 2017-11-04 on ESP32 board with ESP32
Type "help()" for more information.
>>> help('modules')
__main__          heapq             socket            upip
_onewire          io                ssd1306           upip_utarfile
_thread           json              ssl               upysh
array             logging           struct            urandom
binascii          machine           sys               ure
builtins          math              time              urequests
cmath             microWebSrv       tpcalib           uselect
collections       microWebTemplate  ubinascii         usocket
display           micropython       ucollections      ussl
ds18x20           network           uctypes           ustruct
errno             onewire           uerrno            utime
framebuf          os                uftpserver        utimeq
freesans20        pye               uheapq            uzlib
functools         random            uio               writer
gc                re                ujson             ymodem
hashlib           select            uos               zlib
Plus any modules on the filesystem
>>>
````
Here is the output of help('modules') for Loboris_Micropython 3.1.0
````
MicroPython ESP32_LoBo_v3.1.0 - 2017-01-03 on ESP32 board with ESP32
Type "help()" for more information.
>>> help('modules')
__main__          io                ssl               upysh
_thread           json              struct            urandom
array             logging           sys               ure
binascii          machine           time              urequests
btree             math              tpcalib           uselect
builtins          microWebSrv       ubinascii         usocket
cmath             microWebTemplate  ucollections      ussl
collections       micropython       uctypes           ustruct
display           network           uerrno            utime
errno             os                uftpserver        utimeq
framebuf          pye               uheapq            uzlib
freesans20        random            uio               websocket
functools         re                ujson             writer
gc                select            uos               ymodem
hashlib           socket            upip              zlib
heapq             ssd1306           upip_utarfile
Plus any modules on the filesystem
>>>
````

bdespatis, 2018-01-07:
Those modules are frozen modules. Boris can confirm this information.

I still have my old folder and this would be the file to add to your modules folder (components\micropython\esp32\modules
[onewire.py.zip](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/1609928/onewire.py.zip)

We could have them at another folder so people can drag and drop what is needed. A more detailed documentation could be done also on how to build your own firmware. I am not fully capable of changing all I want for now, but soon.

Again, Thanks Boris for all your work !

Regards,
Benoit Despatis, P.Eng.

bdespatis, 2018-01-07:
I checked again, and you can see them at components\micropython\drivers\onewire

Also at components\micropython\esp32\modules_examples\ds18x20
[ds18x20.py.zip](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/1609938/ds18x20.py.zip)

hargathor, 2018-01-08:
Hi @bdespatis 

Thanks for the reply and for the files !

I tried to add other module files to this folder but the version wasn't good since I was missing other dependencies that I couldn't find.
But I have nothing under the path you mentionning
```bash
$ ls -1 components/micropython/esp32/modules_examples/
bme280.py
lora
main.py
mqtt_example.py
ssd1306_i2c_example.py
ssd1306_spi_example.py
tft
thread_example.py
webserver
````
```bash
$ ls -1 components/micropython/drivers/
cc3000
cc3100
display
memory
nrf24l01
README.md
sdcard
wiznet5k
````
Maybe I can try to patch the documentation by explaining the how-to add other modules to the firmware ? I can try if it's ok.

What do you want to say by frozen modules ?

Regards
Romain

bdespatis, 2018-01-09:
Try import machine. I will check tomorrow if I can compile it.

Frozen module 

https://learn.adafruit.com/micropython-basics-loading-modules/frozen-modules

hargathor, 2018-01-09:
Thanks for the link I wasn't aware of what "frozen modules" were. It's really interesting. 

I will build the firmware and flash it. I don't have a ds18b20 or another onewire sensor to test it right now but I will try to import module by repl just to check if everything is ok. I'm a little concerned about an import inside the onewire (_onewire)  that I don't see anywhere. Actually this was my previous issue when I was testing with others module files (that's what I was talking about previously)

hargathor, 2018-01-09:
As expected the `_onewire` module is missing. No issue for the ds18x20 though.
I will try to check into the old files I've got from this repository before the big update.

```bash
>>> import onewire
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "onewire.py", line 5, in <module>
ImportError: no module named '_onewire'
>>> import ds18x20
>>>
````

bdespatis, 2018-01-09:
Hi,

No, it is not missing, but at a different place.
machine.Onewire

You can import machine and create machine.Onewire OR
you can : from machine import Onewire

`>>> help('modules')
__main__          json              ssl               upip_utarfile
_thread           logging           struct            upysh
array             machine           sys               urandom
binascii          math              time              ure
builtins          microWebSrv       tpcalib           urequests
cmath             microWebTemplate  ubinascii         uselect
collections       micropython       ucollections      usocket
display           network           uctypes           ussl
errno             os                uerrno            ustruct
freesans20        pye               uftpserver        utime
functools         random            uheapq            utimeq
gc                re                uio               uzlib
hashlib           select            ujson             writer
heapq             socket            uos               ymodem
io                ssd1306           upip              zlib
Plus any modules on the filesystem
>>> import machine
>>> help(machine)
object <module 'umachine'> is of type module
  __name__ -- umachine
  mem8 -- <8-bit memory>
  mem16 -- <16-bit memory>
  mem32 -- <32-bit memory>
  freq -- <function>
  reset -- <function>
  unique_id -- <function>
  idle -- <function>
  deepsleep -- <function>
  wake_reason -- <function>
  wake_description -- <function>
  heap_info -- <function>
  nvs_setint -- <function>
  nvs_getint -- <function>
  nvs_setstr -- <function>
  nvs_getstr -- <function>
  nvs_erase -- <function>
  nvs_erase_all -- <function>
  loglevel -- <function>
  redirectlog -- <function>
  restorelog -- <function>
  LOG_NONE -- 0
  LOG_ERROR -- 1
  LOG_WARN -- 2
  LOG_INFO -- 3
  LOG_DEBUG -- 4
  LOG_VERBOSE -- 5
  stdin_get -- <function>
  stdout_put -- <function>
  disable_irq -- <function>
  enable_irq -- <function>
  time_pulse_us -- <function>
  random -- <function>
  Timer -- <class 'Timer'>
  Pin -- <class 'Pin'>
  Signal -- <class 'Signal'>
  TouchPad -- <class 'TouchPad'>
  ADC -- <class 'ADC'>
  DAC -- <class 'DAC'>
  I2C -- <class 'I2C'>
  PWM -- <class 'PWM'>
  SPI -- <class 'SPI'>
  UART -- <class 'UART'>
  RTC -- <class 'RTC'>
  Neopixel -- <class 'Neopixel'>
  DHT -- <class 'DHT'>
  Onewire -- <class 'Onewire'>
`

Regards,
Benoit

bdespatis, 2018-01-09:
Can you tell me how to paste the lines like you do ?

Benoit

hargathor, 2018-01-09:
I updated the onewire file to match the correct import. I switch the import `_onewire as _ow` to `from machine import Onewire as _ow` on line 5 to try to keep compatibility. I re-build and reflash the firmware.
I tried the import and everything went well. I stumble upon a new issue while scanning but it may be related to the fact that no onewire device are connected.
I will try later with an actual device and open a new issue if needed since it's more related to the implementation now.

To paste the line like I do I use the backtick, 3 of them followed by the keyword bash and I close my quote using 4 backtick.
[Github - Quoting code](https://help.github.com/articles/basic-writing-and-formatting-syntax/#quoting-code)

Thanks a lot for the support :+1: !

loboris, 2018-01-10:
Sorry for not updating the new Onewire/DS18xx module documentation on time.
I'll try to put the complete documentation on Wiki pages later today and tomorrow.

hargathor, 2018-01-15:
Hi Loboris and first thanks a lot for your work!

Do you had the time to update the wiki regarding the new Onewire library ? I still have trouble importing and using it.

loboris, 2018-01-15:
Sorry, I've been quite bussy during the last week. I expect to finish all the documentation in a next couple of days.

loboris, 2018-01-27:
[Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/onewire) updated

closed 2018-01-27 by loboris