loboris/MicroPython_ESP32_psRAM_LoBo #6: network connecting to non-existent wifi hangs

bwhitman, 2018-01-08:
This worked in the previous version, but since the big update hangs with a stack overflow:

`MicroPython ESP32_LoBo_v3.1.1 - 2017-01-07 on ESP32 with ESP32
Type "help()" for more information.
>>> sta_if = network.WLAN(network.STA_IF)
>>> sta_if.active(True)
True
>>> sta_if.connect("nonexistent","nothing")
>>> no AP found
no AP found
***ERROR*** A stack overflow in task eventTask has been detected.
abort() was called at PC 0x40090c90 on core 0

Backtrace: 0x40090b84:0x3ffc93b0 0x40090c77:0x3ffc93d0 0x40090c90:0x3ffc93f0 0x4008d584:0x3ffc9410 0x4008f1f8:0x3ffc9430 0x4008f1ae:0x00000000

CPU halted.`

Works fine with an actual SSID, and worked fine without an actual SSID in previous version.

bdespatis, 2018-01-09:
Hi,

I don't have the same problem with a DevKitC and a default menuconfig. Did you change something ?

Regards,
Benoit

bwhitman, 2018-01-09:
Hm, yes, some sdkconfig changes. Let me try with a clean one and see if it's still happening.

loboris, 2018-01-18:
@bwhitman
Please reopen if you still have issues.

closed 2018-01-18 by loboris