loboris/MicroPython_ESP32_psRAM_LoBo #8: time.ticks_ms, us return negative values

bboser, 2018-01-08:
The new 64-bit versions of ticks_ms and ticks_us return negative values. 

This is inconvenient in code that checks for a period to expire. E.g.

`python
import time

last_time = 0

while True:
    # do this every 100ms
    while (time.ticks_ms()-last_time) < 100:
        # let's wait until 100ms passed
        pass
    last_time = time.ticks_ms()
    print("do this every 100ms!")
`

Possible fixes are to initialize last_time to a very large negative value, or compute the abs value of the time difference in the inner while loop. 

Still, it would be better if ticks_XX always returned positive values (consistent with standard MicroPython).

loboris, 2018-01-10:
I'll try to fix it.

You should use **time.ticks_diff(start_time, end_time)** to get the time difference:
```
last_time = time.ticks_ms()

while True:
    # do this every 100ms
    while (time.ticks_diff(time.ticks_ms(), last_time)) < 100:
        # let's wait until 100ms passed
        pass
    last_time = time.ticks_ms()
    print("do this every 100ms!")

```

loboris, 2018-01-10:
It is fixed now.
Will be pushed with the next update (later today or tomorrow).

bboser, 2018-01-12:
Thanks, confirm that ticks_xx no longer return negative values.

closed 2018-01-12 by bboser