loboris/MicroPython_ESP32_psRAM_LoBo #10: 'component-quickmail-build' failed

straga, 2018-01-11:
It normal ?
```
/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/quickmail/./quickmail.c:29:
/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/./py/qstr.h:42:39: fatal error: genhdr/qstrdefs.generated.h: No such file or directory
compilation terminated.
/esp32/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/component_wrapper.mk:273: recipe for target 'quickmail.o' failed
make[1]: *** [quickmail.o] Error 1
/esp32/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/project.mk:451: recipe for target 'component-quickmail-build' failed
make: *** [component-quickmail-build] Error 2
make: *** Waiting for unfinished jobs....
```

loboris, 2018-01-12:
I can't reproduce this.
Please try to run `./BUILD clean all`

straga, 2018-01-13:
But Build complete.
But message about quickmail.o exist after delete All, Clone from git and  ./BUILD make -v j4
Without -v, when you don't see, what doing "make", all look "ok" after finish.

loboris, 2018-01-18:
I've tried to reproduce this without success.
`quickmail` components builds and `curl.sendmail()` works without issues.

Please, reopen this issue with more details if needed.

closed 2018-01-18 by loboris