loboris/MicroPython_ESP32_psRAM_LoBo #12: network.ftp - not work in moboXterm

straga, 2018-01-11:
Can connect. passive mode.
List put remove not working.

straga, 2018-01-11:
For example works in thread https://github.com/robert-hh/ESP8266-FTP-Server/ftp.py

loboris, 2018-01-15:
See the notes about FTP server limitations on [Wiki page](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/ftpserver).

It was tested on Windows with FileZilla and TotalCommander FTP plugin and works well.
No problems were detected with Linux clients.

I'll try to see what is the problem with mobaXterm when I find some time, at the moment I can only recommend to use some other FTP client.

straga, 2018-01-15:
It same in Pycharm. 
Like many time ago in second repo it work same: not work. Open folder again and again. 
[Video FTP](https://youtu.be/HT4Lggs5bk0)

loboris, 2018-01-22:
At he moment, you should use supported ftp clients.
I'll try to test with other ftp clients and make it work at some later time.

closed 2018-01-22 by loboris

loboris, 2018-02-07:
Ftp server is now updated.
Tested and works with **MobaXterm**.
It should work also with **Pycharm**, but I haven't tested it.

straga, 2018-02-07:
Cool. I will test later. Thanks