loboris/MicroPython_ESP32_psRAM_LoBo #9: os.listdir() BUG

0x1abin, 2018-01-11:
[image]

bboser, 2018-01-12:
Noticed that, too. Oddly does not occur in '.'.

loboris, 2018-01-12:
**spiffs file system** now uses the (modified) esp-idf **spiffs** driver, which can be configured via menuconfig:
`→ Component config → SPIFFS Configuration`
It looks like you have disabled the `Enable directories` option.

You can try to erase the flash with `./BUILD erase`, then flash the firmware.
The filesystem will be formated, and directories should work as expected.

bboser, 2018-01-13:
Just checked - Enable directories is checked:

Is there some other configuration option for this?

Thanks,
Bernhard

loboris, 2018-01-16:
Thank you for reporting this.

It was my fault, I've forgot to include some modifications to esp-idf spiffs driver needed to correctly list the directories.
The update is just commited, please pull the changes, `os.listdir()` now works as it should with subdirectories.

closed 2018-01-17 by loboris