loboris/MicroPython_ESP32_psRAM_LoBo #13: Thread function from class

straga, 2018-01-12:
For example:
`Class my: 
     _inti(self):
     def start(self):
`
my_test = my()
How run in _thread : _thread.start_new_thread("MY", my.start, ()) argument problem.
 
With navite micropython it work:
_thread.start_new_thread(my.start, ())

loboris, 2018-01-12:
Thanks for reporting this.

`_thread.start_new_thread` was checking the 2nd argument to by of type '**function'**

I've updated the _thread module to also allow the **'bound_method'** type, so the class functions can now be used in threads:
```
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import _thread, utime
=== 
=== class my:
===     def __init__(self, name):
===         self.name = name
===     
===     def start(self):
===         print("hi from " + self.name, utime.ticks_ms())
===         utime.sleep(10)
=== 
=== my_test = my("MyName")
=== 
=== _thread.start_new_thread("MY", my_test.start, ())
=== 
=== 
1073490944
>>> hi from MyName 18432
>>> _thread.list()
ID=1073490944, Name: MY, State: running, Stack=4096, MaxUsed=1292, Type: PYTHON
ID=1073447616, Name: MainThread, State: running, Stack=20480, MaxUsed=2128, Type: MAIN
```
The update will be pushed tomorrow.

loboris, 2018-01-13:
Update pushed, see Wiki page for more details.

closed 2018-01-14 by loboris