loboris/MicroPython_ESP32_psRAM_LoBo #14: Telnet in AP

bdespatis, 2018-01-12:
Hi Boris,

You have set the FTP to work for both AP and STA. Can you do the same for Telnet ?

Also, would it be possible to send to the FTP or the Telnet on which WLAN we want it to connect. We cannot have both STA and AP work at the same time and have the FTP and Telnet. That would be great to have this optional argument.

Regards,
Benoit

loboris, 2018-01-12:
It should work with the latest update.

bdespatis, 2018-01-18:
Any chance we could do the same as mDNS and choose which WLAN element should have the FTP or the Telnet if we are in STA+AP mode?

loboris, 2018-01-22:
At the moment simultaneous STA&AP mode (`WIFI_MODE_APSTA`) is not supported in MicroPython. I have some plans to enable it.
WiFi can operate only in STA or AP mode.
Telnet and Ftp servers will use whatever mode is active when it starts, so there is no point in specifying the wifi interface to use.

If the WiFi mode changes (for example, STA stopped -> AP started), Telnet and/or Ftp server will terminate (`network.ftp|telnet.status()` will return False) and must be started again after the new mode is started.

By the way, I've tested ftp&telnet servers in AP mode and they works without problem.

closed 2018-01-27 by loboris