loboris/MicroPython_ESP32_psRAM_LoBo #15: Only 4 PWM channels available

bboser, 2018-01-12:
Note:
The updated machine.PWM module raises "ValueError: out of PWM channels".

This is since the modified code assigns a new timer to every PWM object. Since the ESP32 has only 4 ledc timers but 8 ledc channels, only 4 channels can be used, all with different frequencies and duty cycles. 

Many applications require several PWM channels all at the same frequency but with different duty cycles. Then the full 8 channels can be useful.

For now I do not need more than 4 channels. But ultimately probably a change is in order, and that may break the existing API.

loboris, 2018-01-12:
I'm working on enabling more than 4 channels if some uses the same frequency.
I'm expecting to push the update in a couple of days.

loboris, 2018-01-13:
Update pushed. See also Wiki page for more details.

closed 2018-01-14 by bboser