loboris/MicroPython_ESP32_psRAM_LoBo #18: Flash fails on osx

scattym, 2018-01-14:
Running a ./BUILD.sh fails with the following error:
```
=========================================
Flashing MicroPython firmware to ESP32...
=========================================
Building partitions from /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
MPY /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modules/logging.py
'make flash' FAILED!
```

Running a make flash directly outputs:
```
hermes:MicroPython_BUILD matt$ make flash
/Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkenv.mk:59: warning: undefined variable `HOST_PLATFORM'
/Users/matt/esp/esp-idf/make/component_wrapper.mk:180: warning: overriding commands for target `libmicropython.a'
/Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkrules.mk:132: warning: ignoring old commands for target `libmicropython.a'
/Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkrules.mk:113: warning: undefined variable `MPY_CROSS_FLAGS'
MPY /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modules/logging.py
make[1]: /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/mpy-cross/mpy-cross: No such file or directory
make[1]: *** [/Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/frozen_mpy/logging.mpy] Error 1
make: *** [component-micropython-build] Error 2
```

Could this be related to:
https://github.com/micropython/micropython/issues/2825

```
cd /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/mpy_cross_build/mpy-cross
make
```
Followed by
```
ln -s /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/mpy_cross_build/mpy-cross/mpy-cross /Users/matt/git/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/mpy-cross/
```
Seemed to resolve the problem for me. Maybe I missed a step in the instructions?

loboris, 2018-01-14:
I can't reproduce this.

I've tried the build process on Mac OSX 10.13 with fresh repository clone and all passes as expected:

- clone the repository
- change the working directory to `MicroPython_BUILD`
- run `./BUILD.sh menuconfig`, set your usb port (mine was `/dev/tty.usbserial-BF101`)
- run `./BUILD.sh -j8 all` to build the firmware
- run `./BUILD.sh flash` to flash the firmware

```
Boriss-iMac:esp32 boris$ git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git
Cloning into 'MicroPython_ESP32_psRAM_LoBo'...
...
...
Boriss-iMac:esp32 boris$ cd MicroPython_ESP32_psRAM_LoBo
Boriss-iMac:MicroPython_ESP32_psRAM_LoBo boris$ cd MicroPython_BUILD
Boriss-iMac:MicroPython_BUILD boris$ ./BUILD.sh menuconfig
Removing old tools versions and cleaning build...
unpacking 'esp-idf'
unpacking MacOS/xtensa-esp32-elf
...
...

Boriss-iMac:MicroPython_BUILD boris$ ./BUILD.sh -j8 all

---------------------
MicroPython for ESP32
---------------------

==================
Building mpy-cross
OK.
==================
================================
Building MicroPython firmware...
================================
Restarting build ...
OK.
--------------------------------
Build complete.
You can now run ./BUILD.sh flash
to deploy the firmware to ESP32
--------------------------------

Boriss-iMac:MicroPython_BUILD boris$ ./BUILD.sh flash

---------------------
MicroPython for ESP32
---------------------

=========================================
Flashing MicroPython firmware to ESP32...
=========================================
Building partitions from /Users/boris/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
Flashing binaries to serial port /dev/tty.usbserial-BF101 (app at offset 0x10000)...
esptool.py v2.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash params set to 0x0230
Compressed 19520 bytes to 11500...
Wrote 19520 bytes (11500 compressed) at 0x00001000 in 0.2 seconds (effective 982.1 kbit/s)...
Hash of data verified.
Compressed 144 bytes to 69...
Wrote 144 bytes (69 compressed) at 0x0000f000 in 0.0 seconds (effective 63.8 kbit/s)...
Hash of data verified.
Compressed 1271408 bytes to 766763...
Wrote 1271408 bytes (766763 compressed) at 0x00010000 in 14.4 seconds (effective 707.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 106...
Wrote 3072 bytes (106 compressed) at 0x00008000 in 0.0 seconds (effective 1158.6 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
OK.

Boriss-iMac:MicroPython_BUILD boris$ 

```

scattym, 2018-01-15:
Yep. Was missing a step. ./BUILD.sh all fixed it up. Thanks!

closed 2018-01-15 by loboris