loboris/MicroPython_ESP32_psRAM_LoBo #19: Can't seem to build esp-idf

Cheaterman, 2018-01-14:
Hi LoBo and thanks for the magic!

I can't seem to build this :-( I tried multiple times and it always fails!

Here is what I get when I do `./BUILD.sh clean all -j5 -v`:
https://gist.github.com/anonymous/9062995b7315e2f87c75ef32a483fc4e

Thanks in advance!

loboris, 2018-01-15:
**The typical steps needed to build the MPy firmware are:**

1. clone the repository, `git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git`
2. change the working directory to `MicroPython_BUILD`
3. run `./BUILD.sh menuconfig`, set your usb port and other options if needed
4. run `./BUILD.sh -j8 all` to build the firmware
5. run `./BUILD.sh flash` to flash the firmware

If step 4 fails:
* run `./BUILD -v all` it will show the compile progress and end on error. Don't use -jN and -v options together, it will be hard to find the error.
* run `./BUILD.sh -j8 clean all` or `./BUILD.sh -v clean all`, sometimes it may help.

There should be no errors after fresh repository clone. I've tested it on all supported operating systems.

Cheaterman, 2018-01-15:
Maybe my OS isn't supported? I'm using Gentoo.

The error is visible on the logs, but alright, let's do it again without -j5.

Cheaterman, 2018-01-15:
The error is hopefully a bit more obvious now:
https://gist.github.com/b106f103a2d43b884ca7cae19b13f734

I think it might be that mpy isn't ready for GCC5, but I'm not familiar with MicroPython or the ESP toolchain so I'll let you tell me what you think :-)

loboris, 2018-01-15:
I've installed Gentoo in VirtualBox (using the CLI minimal 64-bit **image** from [osboxes](https://www.osboxes.org/gentoo/#gentoo-201703-info)) to try the build on Gentoo.

Only had to install `git` & `pyserial` which were not installed.

I've run the steps mentioned in previous post, and all finished without any error.

- clone the repository, `git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git`
- change the working directory to `MicroPython_BUILD`
- run `./BUILD.sh menuconfig`, set your usb port and other options if needed
- run `./BUILD.sh -j8 all` to build the firmware
- run `./BUILD.sh flash` to flash the firmware

```
osboxes@osboxes ~/esp32 $ git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git
Cloning into 'MicroPython_ESP32_psRAM_LoBo'...
...
Resolving deltas: 100% (436/436), done.

osboxes@osboxes ~/esp32 $ cd MicroPython_ESP32_psRAM_LoBo
osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo $ cd MicroPython_BUILD

osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ uname -a
Linux osboxes 4.9.6-gentoo-r1 #1 SMP Tue Mar 21 15:35:56 GMT 2017 x86_64 Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz GenuineIntel GNU/Linux

osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ ./BUILD.sh menuconfig
Removing old tools versions and cleaning build...
unpacking 'esp-idf'
unpacking Linux/xtensa-esp32-elf
make[1]: Entering directory '/home/osboxes/esp32/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
...
...

osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ ./BUILD.sh -j4 all

---------------------
MicroPython for ESP32
---------------------

==================
Building mpy-cross
OK.
==================
================================
Building MicroPython firmware...
================================
OK.
--------------------------------
Build complete.
You can now run ./BUILD.sh flash
to deploy the firmware to ESP32
--------------------------------

osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ ls -l build/*.bin
-rw-r--r-- 1 osboxes osboxes 1271552 Jan 15 14:15 build/MicroPython.bin
-rw-r--r-- 1 osboxes osboxes    3072 Jan 15 14:14 build/partitions_mpy.bin
-rw-r--r-- 1 osboxes osboxes     144 Jan 15 14:15 build/phy_init_data.bin
osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ 

```
You will probably have to rebuild the kernel to include USB to serial drivers to be able to flash the firmware.

Flash FS image creation also works without problem:

```
osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ ./BUILD.sh makefs

---------------------
MicroPython for ESP32
---------------------

=================
Building mkspiffs
OK.
=================
========================
Creating SPIFFS image...
========================
Making spiffs image; Flash address: 0x150000, Size: 2752 KB ...
...
...
--------------------------
To flash to ESP32 execute:
--------------------------
python /home/osboxes/esp32/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x150000 /home/osboxes/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/spiffs_image.img)
-----------------------------
or execute ./BUILD.sh flashfs
-----------------------------
OK.
```

```
osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ ./BUILD.sh makefatfs

---------------------
MicroPython for ESP32
---------------------

================
Building mkfatfs
OK.
================
=======================
Creating FatFS image...
=======================
Making fatfs image; Flash address: 0x150000, Size: 2752 KB ...
...
...
Image file is written to "/home/osboxes/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/fatfs_image.img"
--------------------------
To flash to ESP32 execute:
--------------------------
python /home/osboxes/esp32/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x150000 /home/osboxes/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/fatfs_image.img)
--------------------------------
or execute ./BUILD.sh flashfatfs
--------------------------------
OK.

osboxes@osboxes ~/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD $ 
```

Cheaterman, 2018-01-15:
Hi again LoBo,

Thanks for taking the time to do that! It is very likely that a Gentoo image from March 2017 is using GCC 4.x.

Can you please double-check that for me? If this is correct, it might point to an incompatibility with GCC 5.

Can you also please post the output of `./BUILD.sh -v` so that I can compare the `warning`s that I am getting on my log? If they are absent on your side, it might further indicate a GCC incompatibility.

Thanks again for your help (very appreciated!), I'll get in touch with GCC/Gentoo friends to try and find out what specifically would be needed.

Cheaterman, 2018-01-15:
Hi again LoBo,

I just confirmed the bug on upstream `mpy`. The good news is, well, it's not your fault :-) I'll go make an issue on their repository.

It would still be appreciated that you post your GCC version though :-)

Thanks again!

Cheaterman, 2018-01-15:
Hi again LoBo,

Apparently I'm hitting https://github.com/micropython/micropython/issues/3526 :-(

At least a workaround is included.

Thanks again for all your help and patience!

loboris, 2018-01-15:
The gcc version is  4.9.4
```
osboxes@osboxes ~ $ gcc -v
Using built-in specs.
COLLECT_GCC=/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.4/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/4.9.4/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-4.9.4/work/gcc-4.9.4/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.4 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.4 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.4/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.4/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.4/include/g++-v4 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/4.9.4/python --enable-languages=c,c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 4.9.4 p1.0, pie-0.6.4' --enable-libstdcxx-time --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-altivec --disable-fixed-point --enable-targets=all --disable-libgcj --enable-libgomp --disable-libmudflap --disable-libssp --disable-libcilkrts --enable-vtable-verify --enable-libvtv --enable-lto --without-cloog --enable-libsanitizer
Thread model: posix
gcc version 4.9.4 (Gentoo 4.9.4 p1.0, pie-0.6.4) 
osboxes@osboxes ~ $ 
```
The linked issue has nothing to do with ESP32 build, it is related to pyboard build which has ARM M4 CPU (STM32F405).

I'm upgrading my Gentoo instalation to the newest gcc version, I'll report after testing.

Cheaterman, 2018-01-15:
Thanks for taking the time to upgrade your own GCC for checking, it's really appreciated. Hopefully you come back with a positive result, otherwise we need to find another possible reason :-(

loboris, 2018-01-15:
I've upgraded the whole Gentoo system, gcc version is now 6.4.0
```
osboxes@osboxes ~/esp32/MicroPython/MicroPython_BUILD $ cat /etc/gentoo-release
Gentoo Base System release 2.4.1

osboxes@osboxes ~/esp32/MicroPython/MicroPython_BUILD $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/6.4.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-6.4.0-r1/work/gcc-6.4.0/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/6.4.0 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/6.4.0/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/6.4.0 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/6.4.0/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/6.4.0/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/6.4.0/include/g++-v6 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/6.4.0/python --enable-languages=c,c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 6.4.0-r1 p1.3' --disable-esp --enable-libstdcxx-time --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-altivec --disable-fixed-point --enable-targets=all --disable-libgcj --enable-libgomp --disable-libmudflap --disable-libssp --disable-libcilkrts --disable-libmpx --enable-vtable-verify --enable-libvtv --enable-lto --without-isl --enable-libsanitizer --enable-default-pie --enable-default-ssp
Thread model: posix
gcc version 6.4.0 (Gentoo 6.4.0-r1 p1.3) 
osboxes@osboxes ~/esp32/MicroPython/MicroPython_BUILD $ 
```

I've repeated all the building steps and result is still the same, **no errors.**

Cheaterman, 2018-01-15:
Thanks a lot for your tests, apparently the issue might not be present on GCC 6 (see https://github.com/micropython/micropython/issues/3526 ) so I'll give it a try.

Thanks again!

Cheaterman, 2018-01-15:
Well, GCC 6.4.0 doesn't fix the issue, I'll get in touch with Gentoo/GCC communities. Thanks for everything!

loboris, 2018-01-15:
As I've already mentioned, cross compiling for ESP32 (Xtenssa LX6 microprocessor ) does not depent in any way on the gcc installed on your operating system, nor on gcc ARM M4 compiler .

Cheaterman, 2018-01-16:
Yes, but the tool used for the cross-compilation (`mpy-cross`) is compiled by my system OS, right? And `mpy-cross` seems to segfault, so that's what I'm after. Apparently, it might be related to the fact that I'm using Gentoo Hardened which enables PIC/PIE by default? I tried using `CFLAGS_MOD="-fno-pie -U_FORTIFIED_SOURCE"` but it didn't help (I'm still failing on `*** invalid %N$ use detected ***` despite undefining `_FORTIFIED_SOURCE`).

loboris, 2018-01-16:
You can try to use my precompiled (on Gentoo) `mpy-cross`.
Put it in directory
`MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/mpy-cross`.
If found, the build script will not try to recompile it.
[mpy-cross](http://loboris.eu/ESP32/gentoo/gcc4.9/mpy-cross) or [mpy-cross](http://loboris.eu/ESP32/gentoo/gcc6.4/mpy-cross)

Cheaterman, 2018-01-16:
@loboris yay I finally found it! It was SSP, not PIE! `export CFLAGS_MOD='-fno-stack-protector'`fixes the issue.

I'm not sure that's the right way to go for distribution though? But at least I can compile it myself, yay :-)

loboris, 2018-01-17:
`-fno-stack-protector` is now added to **mpy-cross** [Makefile](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/mpy_cross_build/mpy-cross/Makefile#L35)

closed 2018-01-18 by loboris

Cheaterman, 2018-01-18:
Thanks a lot for everything :-)