loboris/MicroPython_ESP32_psRAM_LoBo #20: Any plans for adding uasyncio?

stroobandt, 2018-01-15:
Are there any plans to add `uasyncio` in the mix?
ESP32 impediments have [recently been resolved](https://github.com/pycom/pycom-micropython-sigfox/issues/112).

loboris, 2018-01-15:
You can add uasyncio as a module (or frozen module).

At the moment, all my tests shows that using threads works better than uasyncio, so, for the moment, I don't have plans to include it as a standard (frozen) module.

stroobandt, 2018-01-15:
Thanks, that is all very useful information for a newbee like myself.
Note to other newbees:
Here is a good [explanation about frozen modules](https://learn.adafruit.com/micropython-basics-loading-modules/frozen-modules).

closed 2018-01-18 by loboris

beyonlo, 2018-12-07:
@loboris Why threads works better than uasyncio? I'm not using MicroPython yet, I just use asyncio for all my applications using CPython (x86/arm) on GNU/Linux, and works like as threads, but I prefer asyncio becouse it is simpler to write applications compared to threads. So, why in the MicroPython threads are better than asyncio?

Ps: in the Python world asyncio is increasingly being used, so much. So, will be a great news for Python programmers to include uasyncio in the standard. :)

Thank you.