loboris/MicroPython_ESP32_psRAM_LoBo #23: ntp_sync() stack overflow with v3.1.4

BluSpanner, 2018-01-16:
Getting a stack overflow when using ntp_sync() with v3.1.4.

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='3.1.4', version='ESP32_LoBo_v3.1.4 on 2017-01-14', machine='ESP32 board with ESP32')
>>> import machine
>>> rtc = machine.RTC()
>>> rtc.ntp_sync('')
>>>
>>> ***ERROR*** A stack overflow in task SNTP has been detected.
abort() was called at PC 0x4008d0b8 on core 0

Backtrace: 0x4008cfd4:0x3ffc2bf0 0x4008d09f:0x3ffc2c10 0x4008d0b8:0x3ffc2c30 0x40089d9b:0x3ffc2c50 0x4008b974:0x3ffc2c70 0x4008b92a:0x00000000

CPU halted.
```

loboris, 2018-01-16:
I cannot reproduce this, have you changed anything in the configuration?
```
>>> rtc = machine.RTC()
>>> rtc.ntp_sync('')
>>> rtc.synced()
True
>>> print("Time set:", utime.strftime("%c"))
Time set: Tue Jan 16 10:35:43 2018
>>> 

```

BluSpanner, 2018-01-16:
Only change the serial port & baud rate

`git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git`
`cd MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD`
`./BUILD.sh menuconfig`
set serial port to /dev/ttyS9
set serial baud to 115200, (max baud rate of LOLIN32 Lite)

`$ diff sdkconfig.old sdkconfig`
```
34,35c34,35
< CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
< CONFIG_ESPTOOLPY_BAUD_115200B=
---
> CONFIG_ESPTOOLPY_PORT="/dev/ttyS9"
> CONFIG_ESPTOOLPY_BAUD_115200B=y
37c37
< CONFIG_ESPTOOLPY_BAUD_921600B=y
---
> CONFIG_ESPTOOLPY_BAUD_921600B=
41c41
< CONFIG_ESPTOOLPY_BAUD=921600
---
> CONFIG_ESPTOOLPY_BAUD=115200
129c129,143
< CONFIG_MICROPY_USE_MQTT=
---
> CONFIG_MICROPY_USE_MQTT=y
>
> #
> # MQTT Configuration
> #
> CONFIG_MQTT_PROTOCOL_311=y
> CONFIG_MQTT_PRIORITY=5
> CONFIG_MQTT_BUFFER_SIZE_BYTE=256
> CONFIG_MQTT_MAX_PAYLOAD_SIZE=2048
> CONFIG_MQTT_LOG_LEVEL=1
> CONFIG_MQTT_LOG_LEVEL0=
> CONFIG_MQTT_LOG_LEVEL1=y
> CONFIG_MQTT_LOG_LEVEL2=
> CONFIG_MQTT_LOG_LEVEL3=
> CONFIG_MQTT_LOG_LEVEL4=
$
```

`./BUILD.sh -j3 clean all`
`./BUILD.sh flash`

loboris, 2018-01-16:
Could you try to flash the prebuilt firmware from `MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/firmware/esp32` and check if it works.
The latest prebuilt firmwares (updated today) have mqtt enabled by default, so it should be the same as the one you've built.
Change the port and baud rate in `flash.sh` first.

BluSpanner, 2018-01-16:
Flashed via

`$ cd MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/firmware/esp32`
`$ esptool.py --chip esp32 --
port /dev/ttyS9 --baud 115200 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_f
req 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0xf000 phy_init_data.bin 0x10000 MicroPython.bin 0
x8000 partitions_mpy.bin`

Still no change

```
>>> import machine
>>> rtc = machine.RTC()
>>> rtc.ntp_sync('')
>>> rtc.synced***ERROR*** A stack overflow in task SNTP has been detected.
abort() was called at PC 0x4008d0b8 on core 0

Backtrace: 0x4008cfd4:0x3ffc2c70 0x4008d09f:0x3ffc2c90 0x4008d0b8:0x3ffc2cb0 0x40089d9b:0x3ffc2cd0 0x4008b974:0x3ffc2cf0 0x4008b92a:0x00000000

CPU halted.
```

loboris, 2018-01-16:
It is strange, I've never had this issue.
Could you try to increase the SNTP task stack size, just edit the file
[MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/machine_rtc](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/machine_rtc.c#L291)
and change in line **291** `512` to `768`.
If it helps I'll set it as default.

BluSpanner, 2018-01-17:
Yes, that has resolve this issue!
Thank you.

closed 2018-01-18 by loboris