loboris/MicroPython_ESP32_psRAM_LoBo #24: network.LAN support

elektrobil, 2018-01-17:
Hi lobo, first of all thank you so much 

New firmware doesn't have network.LAN module or I can't activate. can you help me about this issue

network.LAN(id = None, mdc = Pin(23), mdio = Pin(18), power = Pin(14), phy_addr = 1, phy_type = network.PHY_LAN8720)

>>AttributeError: 'module' object has no attribute 'LAN'

Thanks for your help king regards

loboris, 2018-01-17:
Lan module is not enabled by default.
To enable it, check `→ MicroPython → Modules → Use Ethernet module` in **menuconfig**.

elektrobil, 2018-01-18:
Hi Lobo, Sorry for my English it was selected as you told but LAN class still doesn't appear. Im using LAN8720 chip and using this command before prev version and it has been worked: network.LAN(id = None, mdc = Pin(23), mdio = Pin(18), power = Pin(14), phy_addr = 1, phy_type = network.PHY_LAN8720)

[image]
[image]

loboris, 2018-01-18:
Sorry, my fault, some wrong `#ifdef`s were used.
It is fixed now and committed.

```
>>> import network
>>> from machine import Pin
>>>> 
>>> dir(network)
['__name__', '__init__', 'WLAN', 'phy_mode', 'LAN', 'mqtt', 'telnet', 'ftp', 'mDNS', 'STA_IF', 'AP_IF', 'MODE_11B', 'MODE_11G', 'MODE_11N', 'AUTH_OPEN', 'AUTH_WEP', 'AUTH_WPA_PSK', 'AUTH_WPA2_PSK', 'AUTH_WPA_WPA2_PSK', 'AUTH_MAX', 'PHY_LAN8720', 'PHY_TLK110']
>>> lan=network.LAN(id = None, mdc = Pin(23), mdio = Pin(18), power = Pin(14), phy_addr = 1, phy_type = network.PHY_LAN8720)
D (112628) intr_alloc: Connected src 38 to int 9 (cpu 1)
>>> dir(lan)
['active', 'isconnected', 'status', 'ifconfig']
>>> 
```
I dont'n have the hardware to test it, please report.

elektrobil, 2018-01-18:
I pulled and build but this error appears. What can I do fix this error? (I tried to close the watchdog and still same error appeared)
[image]

loboris, 2018-01-18:
Sorry, I can't test it until I receive the hardware.
The watchdog timeouts should not impact the operation, they just prints the warning.
You can try to disable all watchdogs:
`→ Component config → ESP32-specific → Watch CPU0 Idle Task`
`→ Component config → ESP32-specific → Watch CPU1 Idle Task`
`→ MicroPython → System settings → Enable WatchDog for MicroPython task`
to see if it helps
I don't get the wdt timeouts with no hardware connected.

elektrobil, 2018-01-18:
I checked the everything thats fine  ITS WORK I got the IP via LAN thank you so much but watchdog still have some problems I disabled all watchdogs and idle tasks. 

King Regards

loboris, 2018-01-18:
Great!
I'll try to find what part of the code causes the wdt timeout.

carterw, 2018-01-19:
Seeing that here too. It seems like maybe there is something that should be resetting the timer. Not causing anything to break though.

closed 2018-02-04 by elektrobil