loboris/MicroPython_ESP32_psRAM_LoBo #29: RTC irq module missing?

carterw, 2018-01-19:
Should be able to enter deep sleep and wake up on a timer?
http://docs.micropython.org/en/v1.9.2/esp8266/esp8266/tutorial/powerctrl.html

>>> import machine
>>> rtc = machine.RTC()
>>> dir(rtc)
['init', 'now', 'ntp_sync', 'ntp_state', 'synced', 'wake_on_ext0', 'wake_on_ext1', 'write', 'read', 'clear', 'write_string', 'read_string']
>>> rtc.irq(trigger=rtc.ALARM0, wake=machine.DEEPSLEEP)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'RTC' object has no attribute 'irq'

gepd, 2018-01-19:
Look this example:

```py
import machine

rtc = machine.RTC()

rtc.init((2017, 6, 12, 14, 35, 20))

rtc.now()

rtc.ntp_sync(server="<ntp_server>" [,update_period=])

# <ntp_server> can be empty string, then the default server is used ("pool.ntp.org")

rtc.synced()
# returns True if time synchronized to NTP server

rtc.wake_on_ext0(Pin, level)
rtc.wake_on_ext1(Pin, level)
# wake up from deepsleep on pin level

machine.deepsleep(time_ms)
machine.wake_reason()
# returns tuple with reset & wakeup reasons
machine.wake_description()
# returns tuple with strings describing reset & wakeup reasons

```

carterw, 2018-01-19:
That works, thanks for the tip!

closed 2018-01-19 by carterw