loboris/MicroPython_ESP32_psRAM_LoBo #31: Raspberry Pi Setup Contribution

bdespatis, 2018-01-20:
Hi,

I would like to write about how to configure a Raspberry Pi to use your firmware. How can I contribute to your Wiki ?

As soon as I finish and it's working I will post a link for the binary toolchain (latest version to match yours) and a step by step from a brand new RPi3.

Regards,
Benoit

bdespatis, 2018-01-21:
So far,

I was able to build the toolchain in less than 2 hours (with a fan),
I can erase the flash
I can build the firmware (Took a lot of time, but I did not use -j4)
I can flash the firmware

I cannot use monitor
Exception in thread Thread-1
Traceback [...] threading.py
[...]
AttributeError: 'Console' object has no attribute 'enc_stdin'

According to another forum it could be a msys toolchain error. I will stop this for today.

Regards,
Benoit

bdespatis, 2018-01-21:
Can't stop thinking about it ...

It's fix now. Monitor Functions works.

So Boris (loboris) as soon as you wish that I write down the procedure, I will (for a RPi3)

Regards,
Benoit

loboris, 2018-01-21:
Great, congratulations!
There was some interest for building on RPi, so I would certenly like to include the toolchain for it.
You can put the toolchain and instructions somewhere, the instructions can be in plain text, I'll prepare it for the Wiki.

The toolchain should be usable also on other ARM fruit-Pi boards, I have many of them and can test it.

I would appreciate if you could share the changes needed to build the toolchain from source, I would like to try it.

Best Regards,
Boris

loboris, 2018-01-23:
Any news/progress on this ?

bdespatis, 2018-01-23:
I will send it by email probably. I am on a business trip (since Friday) and I work from 7 to 22 ... I leave today so I will have time to do it.

loboris, 2018-01-24:
The ARM toolchain is now added to the repository.
Thanks.

closed 2018-01-24 by loboris