loboris/MicroPython_ESP32_psRAM_LoBo #33: SPI SD conflict with display SPI

yoooman, 2018-01-23:
Hi lobo, thank you so much for this for micropython that is best version of ESP32.
My  SPI SD  card used same SPI pinout  with  display SPI ,but different CS pin, it not work togerther well.

```bash
>>> os.mountsd()
D (102335) sdspi_host: sdspi_host_init_slot: SPI2 miso=19 mosi=23 sck=18 cs=4 cd=-1 wp=-1, dma_ch=1
D (102336) spi: SPI1 use gpio matrix.
D (102338) intr_alloc: Connected src 30 to int 9 (cpu 1)
D (102343) spi_master: SPI1: New device added, external CS0, effective clock: 20000kHz
I (102351) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (102361) sdmmc_cmd: sdmmc_card_init
D (102407) sdmmc_cmd: SDHC/SDXC card
D (102407) sdspi_host: data CRC set=1
D (102407) sdmmc_cmd: host_ocr=0x40ff8000 card_ocr=0x80ff8000
D (102409) sdmmc_cmd: sdmmc_card_init: host_ocr=40ff8000, card_ocr=80ff8000
D (102417) sdmmc_cmd: switching to DS bus mode
D (102420) sdspi_host: Setting card clock to 20000 kHz
D (102426) spi_master: SPI1: New device added, external CS0, effective clock: 20000kHz
D (102434) vfs_fat_sdmmc: using pdrv=0
---------------------
 Mode: SPI
 Name: SD   
 Type: SDSC
Speed: default speed (25 MHz)
 Size: 948 MB
  CSD: ver=0, sector_size=512, capacity=1941504 read_bl_len=9
  SCR: sd_spec=2, bus_width=5

>>> 
>>> os.listdir('/sd')
['hk', 'System Volume Information', 'dirtest']
>>> 
>>> tft.clear()
>>> os.listdir('/sd')
D (146528) sdspi_host: poll_data_token: timeout
E (146528) sdspi_host: sdspi_host_start_command: cmd=17 error=0x107
D (146529) sdmmc_cmd: sdmmc_req_run returned 0x107
E (146533) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (146541) diskio_sdmmc: sdmmc_read_blocks failed (263)
D (146547) vfs_fat: vfs_fat_readdir_r: fresult=1
[]
>>> os.listdir('/sd')
['hk', 'System Volume Information', 'dirtest']
>>>
```

loboris, 2018-01-23:
It is an issue with **esp-idf** SDCard driver in SPI mode, I'm looking for the solution.

If possible, use the SDCard in 1-line SD mode.
Only 3 pins are needed: 14 (clk), 15 (cmd) and 2 (D0)
It is better than using SDCard in SPI mode on different pins than display's in which case you need 4 pins.

At the moment there is no other solutions for this issue.

yoooman, 2018-01-24:
1-line SD mode can't change pinout, i see tft driver used the **spi_master_lobo.h**, if possible the SD SPI driver can use it?

loboris, 2018-01-24:
Yes, you must have 3 pins dedicated to sdcard.
It is general esp-idf issue, not related to MicroPython and display driver used.

loboris, 2018-01-28:
I'll close this issue for now and reopen it there will be any progress on it.

closed 2018-01-28 by loboris