loboris/MicroPython_ESP32_psRAM_LoBo #35: License and source code ?

Liberasys, 2018-01-26:
Hello Loboris and congratulations for your work !
I have a question concerning your work. Do you provide the source code ? On which license do you produce your code ? For example for uftpserver, I didn't found the source.

Thank you,
BR

stroobandt, 2018-01-26:
Excellent questions!

Liberasys, 2018-01-26:
Maybe you should think about open source and put a "donate" button.
For information, I am trying different high level programming frameworks for an IoT enabled project for a potential customer (I am looking at ESP32 for beginning). I would rather work with free software. If I find a good working framework that reduce my development time, and if my company can legally donate, I will.

stroobandt, 2018-01-27:
Being closed source means that it is difficult to perform a security audit on this. The firmware could effectively be leaking your application code and keys to an off-site server. (I am not saying this is the case, though!)

The lack of proper licensing terms means that loboris could easily sue you in court when you use his software in a commercial setting. However, this has not kept the company M5Stack from using it.

As long as loboris does not resolve both of these issues one way or another, refraining from using this firmware in a commercial setting is your safest option.

The official MicroPython ESP32 port might have less features, but at least it is completely open source and it is licensed under MIT terms which allows for commercial exploitation.

stroobandt, 2018-01-27:
Interestingly, M5Stack _is_ offering an MIT license on _the_ firmware —for not writing _its_ firmware—. Go figure!

icb-, 2018-01-27:
You might want to familiarize yourself with the project before you start spreading FUD and attacking people who give their software to the world. The module that started this issue, the FTP server, is split between [modnetwork.c](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/modnetwork.c#L686-L825) for the micropython interface into the library itself in [ftp.c](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/libs/ftp.c). Both of these have comments giving them the MIT license.

The only part of this build that I'm able to find that *isn't* open source is the ESP IDF binary blobs. There is a config option to exclude these from your builds, but then you'll lose the functionality they support (Bluetooth and WiFi).

Liberasys, 2018-01-27:
Ok, thank you icb for this information. Thus it is merely a problem of communication about licenses.
As the Loboris firmware use different sources (and blobs), there could be a list of external components and corresponding license (and links ?). Plus, clearly displaying the Loboris codes licensing. This would to perfectly the job. What do you think about Loboris ?

stroobandt, 2018-01-27:
@icb- Point well taken! I redacted my initial comments to remove the unintentional FUD factor, now that you pointed us to the source, which admittedly was buried quite deep.
Just regard this as a public auditing exercise. No bad intentions!

loboris, 2018-01-28:
As far as I know, all the code used in this repository is open source, and there are **no binary blobs** without the source code.
The one exception is the part of the esp-idf (low level WiFi/BT drivers) which is closed source.
It is an "issue" with all esp-idf based application.
The only solution, if you want to make fully open source application (in this case the MicroPython firmware), is to not include/link the binary blobs in the firmware (`→ Component config → ESP32-specific → No Binary Blobs`) in which case Wi-Fi/Bluetooth will not work.

Some parts of the code uses different licensing than MIT.
I'll try to collect all the licensing information for used third-party componnents/libraries and also add the licensing information in all of my source code where it is missing.

I'll make all the licensing information available in a separete directory and on the Wiki pages.

stroobandt, 2018-01-28:
@loboris
Thank you for your clarification. This inspires a lot of confidence.
As you suggest, it would indeed be wise to list the items that are not licensed under the MIT terms in a file in the root, together with their license type.
This would allow users who intend on using your software in a commercial setting, to take appropriate actions: e.g. include the GNU GPL license and publish the GNU GPL affected source code.

tuupola, 2018-01-28:
I would get rid of all [GPL licensed code](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/search?utf8=%E2%9C%93&q=gpl&type=). Due it's viral nature GPL requires changing the license from MIT to GPL for the whole project.

Liberasys, 2018-01-29:
Loboris : thank you for all your work. Clear licensing is not fun to do, bit is critical for a community project.
Thank you for your quick reaction and enhancement proposals ! What you propose would be perfect !

For binary blob, I think that maybe Espressif couldn't do otherway because of last FCC laws about WIFI (disallowing open source code for WIFI controllers).

For information, why did you began to build your own MicroPython for ESP32 ? You made a very well work of integration and development !

loboris, 2018-01-29:
I'll try to update all the licensing information in the following days (weeks), so I'll close this issue for now.

@Liberasys
> why did you began to build your own MicroPython for ESP32 ?

I think it's obvious, MicroPython is great programming environment, and the ESP32 is in many ways different than other platforms for which MicroPython exists, so I wanted to build it a different way. It does not mean that my approach is better, it is just better suited for my needs.

loboris, 2018-01-29:
@Liberasys

I'm designing the hardware and software for various measurement stations (mostly in meteorogy/hydrology) of which some are using ESP32/MicroPython.
I'm doing it both in my spare time and as part of my job.

Liberasys, 2018-01-29:
Ok great :-)
Thank you so much for sharing your work Boris !

stroobandt, 2018-01-29:
@loboris Now, you got me curious. Do you somehow make a living out of this? Or is this a hobby project that somewhat got out of hand (in proportions)?

stroobandt, 2018-01-29:
Oh, this is great the way you can combine your daytime job with your hobby, whilst sharing your excellent firmware with the world. Bless you!

tuupola, 2018-02-06:
Why was this closed? Obviously licensing issues have not been fixed. Quickly browsing more GPL code can be found. For example

https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/components/micropython/esp32/modules_examples/lora

which seems to be taken from

https://github.com/Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266

which is GPL licensed.

loboris, 2018-02-06:
Thanks.
I've added the missing licensing information.

I've removed `random.h/.c` with GPL licensing (it wasn't used anywhere!).
I'm working on refactoring `machine_hw_i2c.c` which contains Pycom's GPL license.
I will add a menuconfig option to exclude `quickmail` from build, and later refacor the code to use only `libcurl` for mail functions.

That should leave only MIT abd Apache licensing in the project.

Liberasys, 2018-02-06:
Thank you very much Boris !

mattytrentini, 2018-02-08:
Thanks @loboris! Updating your port to be fully MIT licensed is hugely beneficial. 

(For reference, this was also discussed in the [MicroPython forum](https://forum.micropython.org/viewtopic.php?f=18&t=3553&start=300), across pages 31-32)

Liberasys, 2018-02-09:
Huge beneficial professional uses...
With clear licensing, your project will gain in use an visibility !
Boris, do you need some help for maintaining/developing your firmware ? Funds, testers, thanks card, anything ? :-)

Liberasys, 2018-02-09:
@loboris 
Is it possible for you to add a note about the license in the README.txt, something like:

This firmware is firstly an integration of several codes, mostly :

loboris, 2018-02-09:
I'll try to keep only two licenses: MIT & Apache License 2.0 (esp-idf).

If any part of the code is licensed under other license (GPL) it will be clearly indicated, and this part of the code will be integrated in a way that it can be **excluded** from the build via **menuconfig** in case someone does not want to have a GPL licensed code in the final firmware.

@Liberasys
The more detailed info about licensing will be added soon in a separate file/directory.

Thanks for the help offer. You can help by testing the code, reporting the issues, making suggestions ...
Of course, some donations are also wellcomeed ...

mattytrentini, 2018-02-09:
IANAL but I am under the impression that even the _presence_ of GPL code in the repository will present problems to some...

Liberasys, 2018-02-09:
@loboris 
> The more detailed info about licensing will be added soon in a separate file/directory.
Great !

> Thanks for the help offer.
You are welcome. You make me saving time for my project, you provide support, it is normal to ask !

> You can help by testing the code, reporting the issues, making suggestions ...
I do already :-)

> Of course, some donations are also wellcomeed ...
I think I would donate some euros as person (your project will help me in my DIY projects).
For the IOT product I am beginning to develop, my customer agrees to give back some founds to open source projects that helps me to go faster (for the moment : MicroPython, loboris Firmware, MicroWebSrv). NB : The project is not officially started (startup in creation), and I am in a early study phase.
In brief, where is the DONATE button ? :-)

RQ : be careful to not mixing your personal interest and your company interest. If you want to receive donations, be clear with your employer on the rules to apply (I don't want that you get problems while putting your firmware in open source !). Furthermore be clear of who has the copyright (you or your company, be clear with your employer... In France developers that works for an employer and at the same time at home have a deal, or it often leads to further problems )

loboris, 2018-02-09:
@mattytrentini 
If the GPL code is not compiled and linked into the firmware, there is no way someone could complain if you are selling the product with that firmware without exposing the full source.

loboris, 2018-02-09:
@Liberasys 
My official job and the company I'm working for has nothing to do with MicroPython development.
The rules of the company are such that can do anything at my free time not releted to the company bussines.
I do some paid projects related to ESP32, but the related code (e.g. SDI-12 driver) is not included in MicroPython code.

Liberasys, 2018-02-09:
@loboris 
Ok, so no interest conflicts in sight, good :-)

loboris, 2018-02-12:
I'll close the issue.
Please reopen if anyone still considers it unresolved.

closed 2018-02-12 by loboris

mattytrentini, 2018-02-12:
I don't want to be _that guy_ ;) but wouldn't it make sense to keep this ticket open until you've at least followed through on your [proposal](https://forum.micropython.org/viewtopic.php?f=18&t=3553&start=360#p25368)? At least [machine_hw_i2c.c](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/machine_hw_i2c.c) is still GPL.

I personally would be happier if this were kept open until all GPL code is removed and there was a `LICENSE.md` in the root of the repo explaining the licensing for the repository. 

My understanding is also [in-line with](https://forum.micropython.org/viewtopic.php?f=18&t=3553&start=360#p25363) @tuupola : You _can't_ apply a MIT/Apache license to this repo if there is any GPL code present...

Sorry to be pedantic. But one day, hopefully soon!, I want to use MicroPython at work but I won't be allowed to touch it if it contains GPL code.

loboris, 2018-02-12:
I've had to do some testing for I2C module, it will be committed soon.

Related to the remaining GPL licensed code, I can only quote myself:

> If someone wants to build completely GPL free firmware, the repository can be forked/cloned, components/quickmail directory deleted (leaving no GPL licensed source in the project) and menuconfig used to exclude the method.
> After building, the resulting firmware will not contain any GPL licensed code.
> 
_I'm sorry but I can't consider having GPL licensed  code in the repository **an issue**, especially if you can build the firmware without it._

I've often heard _"many avoid GPL licensed code like plague"_, and mostly the reasoning behind that is: "We want to use someone else's sources to make a profitable product, and we can't do it because there is GPL license".
I'm well aware that there may be other reasons, that is why I'm committed to keep this repository under MIT and Apache licenses as much as possible.

mattytrentini, 2018-02-13:
I understand your position and thanks for clarifying @loboris. 

I still have concerns; it seems possible that the _presence_ of the GPL code mixed in the repository means that the _whole codebase must be GPL_ (as @tuupola alluded to above). 

To be nitpicky it seems clear that the code comprising a _release_ that contains any GPL code must be _entirely released as GPL_ but it's not clear to me if this requires the whole repo to be implicitly GPL too. Forever (?). IANAL and this all seems tricky even by legal standards.

This is discussed a lot but this is quite a [readable discussion](https://softwareengineering.stackexchange.com/questions/204410/how-are-gpl-compatible-licenses-like-mit-usable-in-gpl-programs-without-being-su) on the topic. 

This concern would mean that there'd be a very grey area if someone were to fork and remove the GPL code; the viral nature of the GPL may mean that this is not legally possible since the whole codebase is implicitly GPL. 

I may be wrong and overly worried.

In any case, if you add a `LICENSE.md` and explain the licensing situation then I agree that this is not an issue and ought to be discussed in the forum.

loboris, 2018-02-13:
I think you are "overly worried".

If your compiled firmware (binary) **does not contain a single byte** compiled from **GPL licensed source** (and it can be easily proved) _it cannot be in violation of GPL license_. I'm shure it is as simple as that.