loboris/MicroPython_ESP32_psRAM_LoBo #37: uftpserver answer to FEAT

Liberasys, 2018-01-26:
FEAT request should be answered differently than just 332. Exemple : "211 no-features\r\n".
Here I am trying to use lftp, and FEAT replies 332. Thus lftp think that AUTH TLS is available and use it without success. Please see https://tools.ietf.org/html/rfc2389.

loboris, 2018-01-28:
uftpserver was updated.
FEAT response is now 502. Please test.

Liberasys, 2018-01-29:
Test KO, same comportment as before

> 220 Micropython FTP Server
> FEAT
> 332 
> AUTH TLS
> 332 

On : MicroPython ESP32_LoBo_v3.1.13 - 2017-01-28 on ESP32 board with ESP32
With : git pull, ./BUILD.sh clean, make, flash
(code update is present : ftp_send_reply(211, "no-features");)
Maybe I am missing something ?

loboris, 2018-01-29:
Sorry, I thought you were talking about `uftpserver` frozen module.
I forgot to fix it also in the integrated ftp server, I'll commit the fix in a couple of hours.

Liberasys, 2018-01-29:
Wohooo, I didn't known that you integrated uftpserver as frozen module... :-)
Thus there are 2 codes for FTP server ?
I think I use integrated FTP server : 
```
def start_ftp_server():
    import network
    if network.ftp.start(user="micro", password="python", buffsize=1024, timeout=300):
        print("FTP server started")
        return True
    else:
        print("Error while starting FTP server")
        return False
```
Is there an interest with having your integrated FTP server plus uftpserver frozen module ?

Other question for MicroWebServer : does the menuconfig option websocket server enables MicroWebSocket.py or an other code you wrote ?

Liberasys, 2018-01-29:
> It is a good example of how relatively complex Python module can work in threads, so I should probably move it to examples folder and exclude from frozen modules.

Yes, ideed, it's a good idea I think

> "Use Websockets" menuconfig option includes the websocket module into build. It is not needed for MicroWebServer nor MicroWebSocket

OK

> By the way, I didn't wrote MicroWebServer, as mentioned in the Wiki.
I've only adapted it for this MicroPython specific thread implementation.

Yes and I was speaking by mail with Jean-Christophe (author of MicroWebSrv) the WE, who kindly answered my newbie questions :-)

> I'm planing to implement a C-based WebServer service similar to how ftp and Telnet are implemented.

Great :-)

closed 2018-01-29 by loboris

Liberasys, 2018-01-29:
It is strange... I saw your changes in code.
I did a git pull, BUILD.sh clean, j4, flash.
Version on ESP32 is MicroPython ESP32_LoBo_v3.1.14 - 2017-01-29 on ESP32 board with ESP32
But I still get the problem. Do you have an idea ?

> 220 Micropython FTP Server
> FEAT
> 332 
> AUTH TLS
> 332

loboris, 2018-01-29:
The error was caused by lftp issuing `FEAT` before login, which was not handled correctly in ftp server.
I've fixed it now and will commit soon.

You can disable `FEAT` in lftp and it should work with current version.
`set ftp:use-feat no`
You shoud also disable SSL, not to try to establish TSL connection:
`set ftp:ssl-allow no`

```
boris@UbuntuMate:$ lftp -d -u micro,python 192.168.0.16
---- Resolving host address...
---- 1 address found: 192.168.0.16
lftp micro@192.168.0.16:~> set ftp:use-feat no
lftp micro@192.168.0.16:~> set ftp:ssl-allow no
lftp micro@192.168.0.16:~> ls
---- Connecting to 192.168.0.16 (192.168.0.16) port 21
<--- 220 Micropython FTP Server
---> USER micro
<--- 331 
---> PASS python
<--- 230 
---> PWD
<--- 257 /
---> PASV
<--- 227 (192,168,0,16,7,232)
---- Connecting data socket to (192.168.0.16) port 2024
---- Data connection established
---> LIST
<--- 150 
drw-rw-r--   1 root  root         0 Jan 29 2018 flash
---- Got EOF on data connection
---- Closing data socket
<--- 226 
lftp micro@192.168.0.16:~> ls /flash
---> PASV
<--- 227 (192,168,0,16,7,232)
---- Connecting data socket to (192.168.0.16) port 2024
---- Data connection established              
---> LIST /flash
<--- 150 
drw-rw-r--   1 root  root         0 Jan 29 2018 flash
---- Got EOF on data connection
---- Closing data socket
<--- 226 
---- Closing idle connection
---> QUIT
<--- 221 
---- Closing control socket
lftp micro@192.168.0.16:~> 

```

Liberasys, 2018-01-29:
> You can disable FEAT in lftp and it should work with current version.
> set ftp:use-feat no
> You shoud also disable SSL, not to try to establish TSL connection:
> set ftp:ssl-allow no
Indeed, -e "set ftp:use-feat no' should do the job.
But being compliant with RFCs is better :-)

Anyway, thanks again Boris ! :-)