loboris/MicroPython_ESP32_psRAM_LoBo #38: ftp server pwd and cwd commands don't work

mkarliner, 2018-01-26:
And assuming I have the right code, I can see why. local cwd is not set for the pwd and I think it should be os.listdir rather than uos. 

`elif command == "PWD":
                        cl.sendall('257 "{}"\r\n'.format(cwd))
                    elif command == "CWD":
                        try:
                            files = uos.listdir(path)
                            cwd = path
                            cl.sendall(msg_250_OK)
                        except:
                            cl.sendall(msg_550_fail)`

loboris, 2018-01-28:
uftpserver is now updated.
Could you test again and report the issues, I haven't tested it on OSX.
BTW, uos and os are the same module.

mkarliner, 2018-01-29:
Bug is still present after git pull, ./BUILD.sh clean, ./BUILD.sh, ./BUILD.sh flash
`bigmac:MicroPython_BUILD mike$ ftp 192.168.10.169
Connected to 192.168.10.169.
220 Micropython FTP Server
Name (192.168.10.169:mike): micro
331
Password:
230
Remote system type is UNIX.
Using binary mode to transfer files.
ftp> pwd
Unable to determine remote directory
ftp>`

Why does this matter? Because various ftp sync mechanisms use these commands

loboris, 2018-01-29:
On raspbery Pi you are starting frp client in **passive** mode (`ftp -p 192.168.10.169`).
You may try to use explicit passive mode option (**-p**) on OSX too, even though on Mac OS ftp client should try to use passive mode by default.
The pwd response from ftp server is `257` `PATH`. As `PATH˛is returned over data channel, it will not work in active mode.

There are no `cwd` command in ftp client. Use `cd` to change remote working directory.

mkarliner, 2018-01-29:
To be clear, pwd on RPi with active mode does not work, with passive mode it does.
pwd on OSX does not work, in either mode. I think it's a fairly basic command to be working.
I realise that it may well be an OSC bug, however, that is not a given, as OSX has had ftp for many years, and I've personally used it with micropython and other ftp servers.

loboris, 2018-01-29:
I've tested the ftp on Mac High Sierra (had to instal inetutils with brew).
Tested in passive mode with debug option:

```
Boriss-iMac:~ boris$ ftp -p -d 192.168.0.16
Connected to 192.168.0.16.
220 Micropython FTP Server
Name (192.168.0.16:boris): micro
---> USER micro
331 
Password: 
---> PASS XXXX
230 
ftp> ls
---> PASV
227 (192,168,0,16,7,232)
---> LIST
150 
drw-rw-r--   1 root  root         0 Jan 29 2018 flash
226 
ftp> pwd
---> PWD
257 /
ftp> cd flash
---> CWD flash
250 
ftp> pwd
---> PWD
257 /flash/
ftp> ls
---> PASV
227 (192,168,0,16,7,232)
---> LIST
150 
-rw-rw-r--   1 root  root       113 Jan 01 1970 boot.py
226 
ftp> disconnect
---> QUIT
221 
ftp> quit
Boriss-iMac:~ boris$ 
```

There was a bug in ftp server preventing the connection on clients which issues `FEAT` command before login. It is fixed now and the update will be committed tomorrow.

closed 2018-02-05 by loboris