loboris/MicroPython_ESP32_psRAM_LoBo #39: MicroWebSrv : update please ?

Liberasys, 2018-01-26:
Hello, sorry, I do a lot of issue entry.
I am discovering your Micropython flavour and it is really interesting !
I found that method IsStarted() is not present in the static module you have done with MicroWebSrv.
Maybe it is just an update to do ?
Anyway thank you for all your work !

loboris, 2018-01-26:
You are welcome to open as much issues as you want, it can only help making this port better.

I'll try to address all the issues next week when imm done with some more urgent matters I'm working on.

Thanks.

Liberasys, 2018-01-26:
Thank you for your openness. No problem, this issue is not urgent :-)

loboris, 2018-01-28:
MicroWebSrv is updated now, WebSockets added, some improvements in thread mode...
Tested with example in [examples directory](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/components/micropython/esp32/modules_examples/webserver)

closed 2018-01-28 by loboris

Liberasys, 2018-01-29:
I confirm, /wstest.html is now OK.
Thank you Loboris !