loboris/MicroPython_ESP32_psRAM_LoBo #40: Makefile:324: recipe for target 'zconf.hash.c' failed

stroobandt, 2018-01-26:
This is my first, failed attempt to build `MicroPython_ESP32_psRAM_LoBo` on Xubuntu LTS 16.04.
I followed all Wiki instructions.

```
$ ./BUILD.sh
make[1]: Entering directory '/home/opt/loboris/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
Makefile:324: recipe for target 'zconf.hash.c' failed
make[1]: Leaving directory '/home/opt/loboris/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
make[1]: Entering directory '/home/opt/loboris/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
Makefile:324: recipe for target 'zconf.hash.c' failed
make[1]: Leaving directory '/home/opt/loboris/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
/home/opt/loboris/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/project_config.mk:20: recipe for target '/home/opt/loboris/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig/mconf' failed

===========================================
Error creatimg 'sdkconfig', cannot continue
===========================================

---------------------
MicroPython for ESP32
---------------------

grep: sdkconfig: No such file or directory
grep: sdkconfig: No such file or directory
grep: sdkconfig: No such file or directory
grep: sdkconfig: No such file or directory
grep: sdkconfig: No such file or directory
'sdkconfig' NOT FOUND, RUN ./BUILD.sh menuconfig FIRST.

'make all' FAILED!
```

loboris, 2018-01-27:
Info about **Prerequisites** is added to the [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/build).

Please try to install the required packages.

stroobandt, 2018-01-27:
@loboris Thank you for adding the prerequisites section to the wiki. All worked now.

closed 2018-01-27 by stroobandt