loboris/MicroPython_ESP32_psRAM_LoBo #41: ILI9341 and invrot parameter with M5Stack

tuupola, 2018-01-28:
I am unable to find the correct `invrot` parameter with M5Stack which uses ILI9341. Closes one to work is `1` but it causes mirrored display which also is portrait instead of landscape.

```python
tft = display.TFT()
tft.init(
    tft.ILI9341,
    width=320,
    height=240,
    mosi=m5stack.TFT_MOSI_PIN,
    miso=m5stack.TFT_MISO_PIN,
    clk=m5stack.TFT_CLK_PIN,
    cs=m5stack.TFT_CS_PIN,
    dc=m5stack.TFT_DC_PIN,
    rst_pin=m5stack.TFT_RST_PIN,
    backl_pin=m5stack.TFT_LED_PIN,
    backl_on=1,
    speed=2600000,
    invrot=1,
    bgr=True
)
```

It has been a while when I last read the specsheet but IIRC mirroring was affected by `MADCTL`.  I had similar problem with MicroPython driver but with trial and error figured out [correct  value](https://github.com/tuupola/micropython-ili934x/blob/master/ili934x.py#L118) for M5Stack being `0x08`.

Am I missing something obvious?

loboris, 2018-01-28:
I've just added the `invrot` value **3** which should work for M5Stack display.
The changes will be pushed in a couple of hours. Please test and report if it works.

tuupola, 2018-01-28:
Awesome, thanks!

loboris, 2018-01-28:
Updated.

thingslu, 2018-01-28:
Thank you both, it works on my M5Stack display as well.
Maybe you can add the initialization for it on the example demo:

`#M5Stack:
tft.init(tft.ILI9341, width=240, height=320, rst_pin=33, backl_pin=32, miso=19, mosi=23, clk=18, cs=14, dc=27, bgr=True, backl_on=1, invrot=3)
`

loboris, 2018-01-28:
@thingslu
Thanks for the report.
I've added the initialization in the example. Will be updated with the next commit.

closed 2018-01-28 by loboris

tuupola, 2018-01-29:
I can also confirm with `invrot=3` and `tft.orient(tft.LANDSCAPE)` everything works as expected. The builtin "MicroPython" splashscreen is displayed as portrait. Is there way to make it landscape or alternatively disable it?

loboris, 2018-01-29:
I could add the `init()` option for initial orientation and _splashscreen_.

tuupola, 2018-01-29:
Something like that would be great.