loboris/MicroPython_ESP32_psRAM_LoBo #44: Populate the WIFI event.

Avi-TelnT, 2018-01-29:
Hi Boris,
Since the **old** MicroPython_ESP32_psRAM_LoBo Issues diapered,
Is the WIFI events/call back had been implemented ? 
To notify user code for: 

> typedef enum {
>     SYSTEM_EVENT_WIFI_READY = 0,           /**< ESP32 WiFi ready */
>     SYSTEM_EVENT_SCAN_DONE,                /**< ESP32 finish scanning AP */
>     SYSTEM_EVENT_STA_START,                /**< ESP32 station start */
>     SYSTEM_EVENT_STA_STOP,                 /**< ESP32 station stop */
>     SYSTEM_EVENT_STA_CONNECTED,            /**< ESP32 station connected to AP */
>     SYSTEM_EVENT_STA_DISCONNECTED,         /**< ESP32 station disconnected from AP */
>     SYSTEM_EVENT_STA_AUTHMODE_CHANGE,      /**< the auth mode of AP connected by ESP32 station changed */
>     SYSTEM_EVENT_STA_GOT_IP,               /**< ESP32 station got IP from connected AP */
>     SYSTEM_EVENT_STA_LOST_IP,              /**< ESP32 station lost IP and the IP is reset to 0 */
>     SYSTEM_EVENT_STA_WPS_ER_SUCCESS,       /**< ESP32 station wps succeeds in enrollee mode */
>     SYSTEM_EVENT_STA_WPS_ER_FAILED,        /**< ESP32 station wps fails in enrollee mode */
>     SYSTEM_EVENT_STA_WPS_ER_TIMEOUT,       /**< ESP32 station wps timeout in enrollee mode */
>     SYSTEM_EVENT_STA_WPS_ER_PIN,           /**< ESP32 station wps pin code in enrollee mode */
>     SYSTEM_EVENT_AP_START,                 /**< ESP32 soft-AP start */
>     SYSTEM_EVENT_AP_STOP,                  /**< ESP32 soft-AP stop */
>     SYSTEM_EVENT_AP_STACONNECTED,          /**< a station connected to ESP32 soft-AP */
>     SYSTEM_EVENT_AP_STADISCONNECTED,       /**< a station disconnected from ESP32 soft-AP */
>     SYSTEM_EVENT_AP_PROBEREQRECVED,        /**< Receive probe request packet in soft-AP interface */
>     SYSTEM_EVENT_GOT_IP6,                  /**< ESP32 station or ap or ethernet interface v6IP addr is preferred */
>     SYSTEM_EVENT_ETH_START,                /**< ESP32 ethernet start */
>     SYSTEM_EVENT_ETH_STOP,                 /**< ESP32 ethernet stop */
>     SYSTEM_EVENT_ETH_CONNECTED,            /**< ESP32 ethernet phy link up */
>     SYSTEM_EVENT_ETH_DISCONNECTED,         /**< ESP32 ethernet phy link down */
>     SYSTEM_EVENT_ETH_GOT_IP,               /**< ESP32 ethernet got IP from connected AP */
>     SYSTEM_EVENT_MAX
> } system_event_id_t;
>

loboris, 2018-01-29:
Thanks for reminding me, I've never put it in the released version.

I've tested it and it should be included with the next commit.
It works in a way that you define a callback function which is executed on wifi event and gets the wifi event number as a parameter.

Avi-TelnT, 2018-01-29:
Great. Thank you Boris,
Is all data will be also with the calledback? 
As in SYSTEM_EVENT_AP_PROBEREQRECVED will be with MAC and RSSI ? 
as 
`SYSTEM_EVENT_AP_PROBEREQRECVED: {
        system_event_ap_probe_req_rx_t *ap_probereqrecved = &event->event_info.ap_probereqrecved;
        ESP_LOGD(TAG, "SYSTEM_EVENT_AP_PROBEREQRECVED, rssi:%d, mac:" MACSTR, \
                   ap_probereqrecved->rssi, \
                   MAC2STR(ap_probereqrecved->mac));
        break;
`

Can you prefect when will be next commit?

loboris, 2018-01-29:
The data can be included in the event callback.

You must be aware that you may lose some events if they occurs too frequently.
MicroPython has 8 items callback queue, and the callback latency can be quite high (up to 10 ms).
Your callback function (Python code) will also need some time to execute, so try to make it as short as possible.

The updated is planned for today.

Avi-TelnT, 2018-01-29:
I believed 10 ms is totaly OK. 
I hope the callback queue will not blocked normal User code. 
And it will be as define event as IRQ, and get the event as IRQ ISR. 
If you can include small Python demo for SYSTEM_EVENT_AP_PROBEREQRECVED RSSI and MAC, I believe I can do the same to all other WIFI events. 
Thank you.

loboris, 2018-01-29:
The update with WiFi events enabled is now commited.

Description and simple example is available on [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/network).

Avi-TelnT, 2018-01-29:
Thank you Boris,
All seems to work on first run. 
I will run more test and report here. 
Great work.

Avi-TelnT, 2018-01-30:
All seems working OK.  AP and STA WiFi events, beside the SYSTEM_EVENT_AP_PROBEREQRECVED event.
According to https://github.com/espressif/esp-idf/issues/742 at Jun 29, 2017 it seems this event has been forbidden in Jun 29, 2017 IDF 
Because it send have too much SYSTEM_EVENT_AP_PROBEREQRECVED 
It state to be enabled after optimize this event in the release next to 2.1.
**Boris,** 
Do you know if it was enabled? 
And if not, where I can raw enable it? 
Raw, means without optimize anything. 
Acording to https://github.com/espressif/esp-idf/blob/3a271a4ae7df8a9049fbbb801feafca5043c31eb/docs/api-guides/wifi.rst
It state that this event will remove totally, which is terrible to my work. on ESP8266 it works fine.

loboris, 2018-01-30:
Nothing has changed with regard to this event as far as know.
All wifi events are generated by the wifi driver and the driver is closed source, so you can't do anything about it.

Avi-TelnT, 2018-01-30:
OK. Thank you.

Avi-TelnT, 2018-01-30:
**Boris,** 
Is esp_wifi_set_promiscuous(bool en) has been populated in your code? 
Maybe I get the AP_PROBEREQRECVED event using this raw API.

loboris, 2018-01-30:
`esp_wifi_set_promiscuous()` is not used anywhere.
I don't think you can get AP_PROBEREQRECVED event if it is disabled in the wifi driver.

Avi-TelnT, 2018-01-30:
It so strange that SYSTEM_EVENT_AP_PROBEREQRECVED works OK on Lua ESP8266 and disable on ESP32 while if faster and more powerful.

loboris, 2018-01-30:
I'm always using the latest esp-idf master branch.
When/if SYSTEM_EVENT_AP_PROBEREQRECVED is enabled you'll get it in MicroPython to.

Avi-TelnT, 2018-01-30:
Can it be that  ESP_LOGD is not working in this last updates? 
I enable it in menuconfig compile and upload all again, but it does not seems to be working

Here is my sdkconfig for referance

# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=y
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=2
CONFIG_BOOTLOADER_VDDSDIO_BOOST=y

#
# Security features
#
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=y
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_ESPTOOLPY_AFTER="no_reset"
CONFIG_MONITOR_BAUD_9600B=
[... 524 lines cut ...]

#
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_SIZE_4096=
CONFIG_WL_SECTOR_SIZE=512
CONFIG_WL_SECTOR_MODE_PERF=
CONFIG_WL_SECTOR_MODE_SAFE=y
CONFIG_WL_SECTOR_MODE=1

loboris, 2018-01-30:
You must also set the log level for MicroPython:
`→ MicroPython → System settings → Set ESP32 log level`

Or, you can set the debug (or any other) log level from MicroPython:
`machine.loglevel("*", machine.LOG_DEBUG)`

The documentation for this will be added to the Wiki soon.

Avi-TelnT, 2018-01-30:
I have this setting 
→ MicroPython → System settings → Set ESP32 log level
"Set ESP32 log level (Debug)"
I also add machine.loglevel("*", machine.LOG_DEBUG) in main.py first line after imports 
**and still not working,**  very odd. 
I will check with the old code without the wifi events

loboris, 2018-01-30:
You have the global esp-idf log level set to ERROR:
`→ Component config → Log output → Default log verbosity`
It must be higher or equal to MicroPython log level.

Avi-TelnT, 2018-01-30:
It was set for ERROR instead of debug, before the last ./BUILD.sh menuconfig
Maybe ./BUILD.sh menuconfig do not save properly. 
The test old code without the wifi events, is **debug is working**,
I see the completion, of old code,  after menuconfig
In the new code, after ./BUILD.sh menuconfig is  very odd 
As it does not compile. No long list of files been compile. 
I did
./BUILD.sh clean
./BUILD.sh all
./BUILD.sh flash
./BUILD.sh flashfs

Avi-TelnT, 2018-01-30:
I set Component config → Log output → Default log verbosity to be "Debug"
Did 
./BUILD.sh clean
./BUILD.sh all
Did not show a list of file been complied. 
It delay on screen, as do that, but now progress, as many file names on screen.

loboris, 2018-01-30:
I don't understand the problem.
`./BUILD.sh` never shows anything except few messages about the process unless you use `-v` option.
It has nothing to do with setting the esp-idf log level.

Avi-TelnT, 2018-01-30:
I never used -v option. 
Old  code show the list of file been compiled when I change log level. 
The last code, it not showing. 
Were you put the -v? 
./BUILD.sh clean
./BUILD.sh all -v

Avi-TelnT, 2018-01-30:
The system crash on newer code, when I use debug level and run your demo wifi event code.
Without debug level set, it was OK. 
I will test more to pinpoint the problem.  

I (197222) wifi: mode : sta (30:ae:a4:22:a4:f4) + softAP (30:ae:a4:22:a4:f5)
I (197223) wifi: pm stop, total sleep time: 0/189236511

W (197225) [Ftp]: FTP RESET
D (197225) event: SY>>> STEM_EVENT_AP_START
D (1***ERROR*** A stack overflow in task Telnet has been detected.
abort() was called at PC 0x4008d184 on core 1
0x4008d184: vApplicationStackOverflowHook at /home/telnt/esp/MicroPython/Tools/esp-idf/components/esp32/./panic.c:648

Backtrace: 0x4008cfc8:0x3ffe3560 0x4008d16b:0x3ffe3580 0x4008d184:0x3ffe35a0 0x40089e0c:0x3ffe35c0 0x4008b9ac:0x3ffe35e0 0x4008b962:0xa5a5a5a5
0x4008cfc8: invoke_abort at /home/telnt/esp/MicroPython/Tools/esp-idf/components/esp32/./panic.c:648

0x4008d16b: abort at /home/telnt/esp/MicroPython/Tools/esp-idf/components/esp32/./panic.c:648

0x4008d184: vApplicationStackOverflowHook at /home/telnt/esp/MicroPython/Tools/esp-idf/components/esp32/./panic.c:648

0x40089e0c: vTaskSwitchContext at /home/telnt/esp/MicroPython/Tools/esp-idf/components/freertos/./tasks.c:4702

0x4008b9ac: _frxt_dispatch at /home/telnt/esp/MicroPython/Tools/esp-idf/components/freertos/./portasm.S:406

0x4008b962: _frxt_int_exit at /home/telnt/esp/MicroPython/Tools/esp-idf/components/freertos/./portasm.S:206

CPU halted.

loboris, 2018-01-30:
`./BUILD.sh -v any_command(s)`

[Build Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/build), BUILD.sh -> Options

It looks like the Telnet server needs a bigger stack when hihger log level is used.
I'll check it and fix it.

loboris, 2018-02-12:
I was unable to reproduce this issue.
Please report if the issue is still present and reopen if needed.

closed 2018-02-12 by loboris