loboris/MicroPython_ESP32_psRAM_LoBo #45: RTC timezone : not applied ?

Liberasys, 2018-01-29:
Hello Loboris,
I set timezone to Europe/Paris in menuconfig.
Time is NTP synced.
```
>>> utime.gmtime()
(2018, 1, 29, 15, 40, 52, 2, 29)
>>> utime.localtime()
(2018, 1, 29, 15, 40, 58, 2, 29)
```
It seems that timezone is not applied (it replies the same between gmttime and localtime, and both are not GMT+1 as they should be in France) ?

loboris, 2018-01-29:
You can now set the time zone when calling `ntp_sync()`

```
>>> rtc = machine.RTC()                                                                                                                     
>>> rtc.ntp_sync(server="hr.pool.ntp.org", tz="CET-1CEST")                                                                                  
>>> rtc.synced()                                                                                                                            
True                                                                                                                                        
>>> utime.gmtime()                                                                                                                          
(2018, 1, 29, 16, 3, 18, 2, 29)                                                                                                             
>>> utime.localtime()                                                                                                                       
(2018, 1, 29, 17, 3, 30, 2, 29)                                                                                                             
>>> 
```

For time zone string use the **2nd** entry from `components/micropython/docs/zones.csv`
"Europe/Paris",`"CET-1CEST,M3.5.0,M10.5.0/3"`
or shorter version: `"CET-1CEST"`

Liberasys, 2018-01-29:
Ok, nice ! It is working, thank you.
A new error spawned (I think that a int cast is lacking...)
`
>>> local_time = utime.mktime(utime.localtime())
>>> print(local_time.strftime("%Y/%m/%d - %H:%M:%S"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'float' object has no attribute 'strftime'

`

loboris, 2018-01-29:
`utime.mktime(tupletime)`

> Accepts an instant expressed as a time-tuple in local time and returns a floating-point value with the instant expressed in seconds since the epoch.

```
>>> utime.mktime(utime.localtime())
1517249920.0
```

`time.strftime(fmt[,tupletime])` 

> Accepts an instant expressed as a time-tuple in local time and returns a string representing the instant as specified by string fmt.
```
>>> utime.strftime("%Y/%m/%d - %H:%M:%S", utime.localtime())
'2018/01/29 - 19:27:57'
>>> utime.strftime("%Y/%m/%d - %H:%M:%S")
'2018/01/29 - 19:29:15'
>>> 
```
Quoted definitions from [tutorialspoint](https://www.tutorialspoint.com/python/python_date_time.htm)

Liberasys, 2018-01-29:
Ok, I think I missed something in documentation...
Thank you for your example Boris !

closed 2018-01-29 by loboris

djairjr, 2018-01-31:
Boris, can you put this on the Wiki?

djairjr, 2018-01-31:
When I said "this", I said this:

> For some time you can set the time zone when calling ntp_sync()

> rtc = machine.RTC()
> rtc.ntp_sync(server="hr.pool.ntp.org", [...]
True                                                                                                                                        
>>> utime.gmtime()                                                                                                                          
(2018, 1, 29, 16, 3, 18, 2, 29)                                                                                                             
>>> utime.localtime()                                                                                                                       
(2018, 1, 29, 17, 3, 30, 2, 29)                                                                                                             
>>> 
For time zone string use the 2nd entry from components/micropython/docs/zones.csv
"Europe/Paris","CET-1CEST,M3.5.0,M10.5.0/3"
or shorter version: "CET-1CEST"

djairjr, 2018-01-31:
How can I help you to update the wiki? I am very interested in your port...

loboris, 2018-02-02:
@djairjr 
You are welcome to post here any correction, suggestion or addition to the Wiki pages.

Liberasys, 2018-02-06:
Hello, just to say that everything works !
./sdkconfig extract : 
CONFIG_MICROPY_TIMEZONE="CET-1CEST,M3.5.0,M10.5.0/3"

REPL test:
```
>>> time.gmtime()
(2018, 2, 6, 13, 3, 14, 3, 37)
>>> import time
>>> time.localtime()
(2018, 2, 6, 14, 6, 2, 3, 37)
>>> time.gmtime()
(2018, 2, 6, 13, 6, 8, 3, 37)
```

On the menuconfig, the help message is:
CONFIG_MICROPY_TIMEZONE: 
  │ Set the default time zone string used when updating the time from NTP server
  │ You can use 'components/micropython/docs/zones.csvzones.csv'
  │ file to find the time zone string.
  │ Use the second entry for your Region/city as the value set here     
Remarks:
- components/micropython/docs/zones.csv is the good path (copy/paste problem I think :-)
- add example : For Zagreb, the zones.csv file contains :"Europe/Zagreb","CET-1CEST,M3.5.0,M10.5.0/3". You should put "CET-1CEST,M3.5.0,M10.5.0/3" as Time zone.

Thank you Boris !

loboris, 2018-02-06:
@Liberasys

Thanks, I've corrected the menuconfig help entry and added the example, will be visible on the next update.

Heve you noticed that you can also specify the timezone when calling `ntp_sync()` ?

`rtc.ntp_sync(server="hr.pool.ntp.org", tz="CET-1CEST")`

Liberasys, 2018-02-07:
@loboris 
OK for menuconfig help.
Short form works too indeed. For my information, what is the part after the 1st coma ?
I confirm that setting timezone when calling ntp_sync() works too.

Thank you boris, everything is good for me at now :-)