loboris/MicroPython_ESP32_psRAM_LoBo #46: DNS server ?

Liberasys, 2018-01-29:
I need to do an IoT project that has a configuration mode. In this mode, the ESP32 would: 
- spawn a WIFI access point (OK),
- serve HTML  for doing configuration of the connected thing - like WIFI (OK),
- act as DNS server (catching all requests from connected WIFI client) (KO).
I have seen that before with ESP8266, is there a possibility to integrate this function ?
I know there is mDNS but I would rather use standard DNS service. Do you think it is overkill ?

loboris, 2018-01-29:
No other DNS implementation apart from mDNS is planned at this time.

Could you be more specific about what do you need/expect from DNS server on ESP32 ? Some example maybe ?

Liberasys, 2018-01-29:
The goal is that in configuration mode, any WIFI client using a web client and asking any website would be redirected to ESP32 webserver IP.
This gives the possibility to have a configuration interface that doesn't require any special software (using AP client and standard browser). 

A full example with "configuration portal mode" AP is available here : https://github.com/zhouhan0126/WIFIMANAGER-ESP32
In this example the author uses the automatic portal procedure available in ios and android in order to get the local WIFI config from the user.

Here is an other sample with "simple configuration portal" : https://gist.github.com/Cyclenerd/7c9cba13360ec1ec9d2ea36e50c7ff77

Here is a library for Arduino : https://github.com/esp8266/Arduino/tree/master/libraries/DNSServer

In a perfect world, the DNS server should either respond good to a list of FQDNS, either to respond good to every FQDNs (see mode in Arduino example :   // if DNSServer is started with "*" for domain name, it will reply with provided IP to all DNS request)

loboris, 2018-01-29:
For the purpose you've described, probably the pure Python implementation could be used.
There are two projects that I know of [uPyPortal](https://github.com/lemariva/uPyPortal) and [MicroDNSSrv](https://github.com/jczic/MicroDNSSrv) which may be interesting.
I'll look at it when I find some time.

Liberasys, 2018-01-30:
Yes of course, some Python code could be used. And thank you for having pointed me those examples !
It is just that base services are included in your firmware thus I was telling myself that it could be the case for DNS server :-)

Liberasys, 2018-01-31:
For information, jczic has patched his code for running under your firmware, see https://github.com/jczic/MicroDNSSrv/issues/2.
It is perfectly suited to my needs.
Thank you.

loboris, 2018-02-02:
Thanks for the information.
I'll probably add MicroDNSSrv as a frozen module after some testing...

closed 2018-02-02 by loboris

Liberasys, 2018-02-04:
Nice, form information JCZic is improving his code at now :-)