loboris/MicroPython_ESP32_psRAM_LoBo #47: Secure boot and flash encryption

alankaranak, 2018-01-31:
Is there any example of working with flash-encryption? For example, I need to write to the internal memory a file that could be read from the python script's, but it's impossible to read it using ampy etc.

Are there any way to protect or close access to the uart console after the device is burned?

loboris, 2018-02-01:
That is not what [Flash encription](http://esp-idf.readthedocs.io/en/latest/security/flash-encryption.html) is meant to be.
Ampy and similar programs accesses the file system the same way as Python script.

You can prevent accessing the system from uart console (REPL) or Telnet simply by not starting the REPL and Telnet.
You can place an infinite loop at the end of the **main.py** and REPL will never be started.
```
...
...
while (True):
    time.sleep(1)

```

alankaranak, 2018-02-13:
I apologize, did not quite understand. If I connect to the console at runtime, I can interrupt the script at any time (ctrl + c). And get access.
I have a task to prevent access through the REPL if someone connected to the console. Then he can see and download any file from the file system:
ampy --port COMX ls / flash

loboris, 2018-02-13:
I Think I understand now.
I will add a method to disable / enable the uart access. The option to enable uart access  by special code sequence could be added if you need it.

alankaranak, 2018-02-13:
OK, thank you very much!

loboris, 2018-03-18:
The method to disable `stdin` (uart access) was added some time ago, sorry for not informing you about that.
You can use `machine.stdin_disable(pattern)` to disable (ignore) the input on uart.
`pattern` is the string of up to 15 characters used to enable the input again. If it is matched, the input will be enabled.
For example:
```python
import machine
machine.stdin_disable("**__77x??")
```
The input from the uart will be disabled.
If the pattern `**__77x??` is sent over the uart, the input will be enabled.

loboris, 2018-03-24:
I'll close the issue, please reopen if needed.

closed 2018-03-24 by loboris