loboris/MicroPython_ESP32_psRAM_LoBo #48: Documentation: Link to module examples from wiki

stroobandt, 2018-01-31:
Hi Boris,
At first, I had overlooked your [module examples](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/components/micropython/esp32/modules_examples), simply because they are not mentioned on the front page of the [wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki). They are only mentioned on the [project code page](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo).
It would be handy to see that link also being mentioned prominently on your wiki page.

loboris, 2018-02-01:
Thanks for the suggestion. I have added the link to the examples on the Wiki menu.

pastukhov, 2018-02-02:
I think, mqtt example is wrong, mqtt.subscribe should be inside connected_cb.

loboris, 2018-02-02:
> I think, mqtt example is wrong, mqtt.subscribe should be inside connected_cb.

Not necessarily.
You can run your mqtt code **without** using any callbacks, except for `data`, they are just an option which can be convenient in some cases.
There is a method `mqtt.status()` which can be used to check the mqtt status.
You can run your mqtt code in a **thread** and use only the `data `callback to set some variable.

Anyway, I agree the mqtt example should be expanded to contain more detailed example of various mqtt usage scenarios. I'll try to do that soon.

closed 2018-02-02 by loboris