loboris/MicroPython_ESP32_psRAM_LoBo #50: Problem with UART :  UART event queue full

Liberasys, 2018-02-02:
Hello,
I am trying to use UART2. For this, I try this code:

elektrobil, 2018-02-02:
Hi Lobo and  Liberasys 
I have same problem too with UART its becouse I dont writre a new issue. Im reading some data with UART but ture data returns only 21 times (21x12=252byte) after 66th cycle returns true data until 88th cycle
`var = uart.read(12)`
How is that possible and Is this the source of the error because of the code I wrote? 
P.S. : This code running in a thread

King Regards

loboris, 2018-02-02:
Thanks for reporting this.
**machine.UART** is now updated, bug fixed and should function as intended.

One new argument is added to the `machine.UART.init()`: `buffer_size`
The default buffer size is 256 bytes and can be now set in the range `256` - `4096`
For example:
`uart = machine.UART(2, tx=22, rx=21, buffer_size=1024)` 

The full documentation for the UART module will be added soon.

elektrobil, 2018-02-02:
Thank you for your update @loboris  but still same error how can I clean the buffer every cycle
`true data = b'2.012/022\x00\r\n'`
after fill the buffer 98th cycle 
`wrong data = b'2/022\x00\r\n3.26'`

I want clean buffer every cycle in my loop

Thank you so much for your help
King regards

loboris, 2018-02-02:
@elektrobil
I'm not shure if I understand the problem.
Maybe you could give more detailes, what method you are using to read, what the other side sends, ...

If you need to clean the input buffer, you should read until nothing is returned.

I'm planning to update the UART module with more methods like input buffer clear, read with timeout, read until a specified pathern, etc.

elektrobil, 2018-02-02:
Hi @loboris thank you for your helps
My project is nrf24le1 reads the sensor and send to the main nrf24le1. Main Nrf chip connected to Esp32 via Uart.

`import machine `

`uart = machine.UART(2, tx=pin1, rx=pin2, buffer_size=1024) `
`uart.init(9600, bits=8, parity=None, stop=1, timeout=10000)`
`var = uart.read(12)`

This code runs in a thread function continious
Everything ok untill buffer size full 
true data = b'2.012/022\x00\r\n'
after fill the buffer 98th cycle (buffer size 1024)(when buffersize =256 cycle return error at 21st cycle)
wrong data = b'2/022\x00\r\n3.26' last 3.26 must be in front of the variable.
I think I need to clean input buffer. Is that possible with micropython code because Im new in micropython
Thank you so much again

loboris, 2018-02-02:
Could you please post your complete thread function. I think there should be a simple solution for that.

elektrobil, 2018-02-03:
Hi @loboris this is my thread function below Thank you for your help
`       gc.collect()`
`        i=0`
`        inx=0`
`        htext = 35`
`        vtext = 10`
`        if uart.any():`
`            gelen = uart.read(12)`
`            l+=1`
`            print(gelen)`
`            print("SAYI = ",l)#count loop cycle`
`            gelens = gelen.decode()`
`            print(len(gelens))`
`            print(gelens)`
`            gelens = gelens.rstrip('\n')`
`            gelens = gelens.rstrip('\r')`
`            gelens = gelens.rstrip('\x00')`
`            if len(gelens) == 9:`
`                gelens,sets = gelens.split('/')`
`                sb = list(sets)`
`                print(sb)`
`                if int(sb[1])>1:`
`                    inx = int(sb[2]) + (4 * (int(sb[1])-1))`
`                else:`
`                    inx = int(sb[2])`
`                #print(inx)`
`                vals[inx-1] = float(gelens)`

King Regards...

loboris, 2018-02-03:
I suppose this code is included in an infinite loop in your thread function ?
If not, it will run only once.

`uart.read(12)` reads 12 bytes from uart buffer and removes them from the buffers, so if it is executed regularly no buffer overflow should occur.
 
You can use `if uart.any() >= 12:` to test if at leas 12 bytes have arrived.

The UART module is updated today, some new functions are added as well as callbacks on some uart events. Full documentation will be on Wiki tomorrow.

Note:
When you insert the code here, paste the code, select all code and press **<>**.
The code should apear betveen two sets of **```**
That way the code structure will be preserved.

elektrobil, 2018-02-03:
I think buffer can't clean and I have the same error. How could I be sure to clean the buffer ?
This sample : using `uart = machine.UART(2, tx=pin1, rx=pin2, buffer_size=256)`
(it seems to me) Buffer Not Full Output:
**b'3.269/023\x00\r\n'**
**SAYI =  21**
12 (chars)
3.269/023 -- (mydata/id)
['0', '2', '3']
-------------------------------------- after 21st cycle it means 21x12 = 252bytes
(it seems to me)Buffer Full Output:
**b'2/022\x00\r\n3.26'**
**SAYI =  22**
12
2/022
3.26

Thank you for help on saturday night

elektrobil, 2018-02-04:
Hi @loboris  I pulled the new version but data comes like that.
```
0;0.0;0.0;0.
b'0;0.0;0.0;0.'
SAYI =  3
```

elektrobil, 2018-02-05:
Thank you @loboris 
UART works good 
King Regards

loboris, 2018-02-05:
Please check the new [UART Wiki page](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/usrt).
There are some new methods, and you can now use callback in your code.
For your application which reads the messages ending with '\r\n' you can use the new method `uart.readln()`.

elektrobil, 2018-02-05:
I sow Wiki page when you commit. Thank you so much. It works perfect

Liberasys, 2018-02-06:
Hello, UART update seems great Boris ! UART callback with pattern is very cool !
I tested with putty as serial terminal client under Linux. uart.readln() works great with lineend='\r'.
Thank you Boris !

closed 2018-02-09 by Liberasys