loboris/MicroPython_ESP32_psRAM_LoBo #55: Build Failing on Windows 10

neoxharsh, 2018-02-04:
Build Failing on Windows 10

```
CC build/mbedtls/library/hmac_drbg.o
CC build/mbedtls/library/memory_buffer_alloc.o
CC build/mbedtls/port/esp_sha512.o
CC build/mbedtls/port/esp_hardware.o
CC build/mbedtls/port/net_sockets.o
CC build/mbedtls/port/esp_bignum.o
CC build/mbedtls/port/esp_sha256.o
CC build/mbedtls/port/esp_sha1.o
CC build/mbedtls/port/mbedtls_debug.o
AR build/mbedtls/libmbedtls.a
CC build/mdns/mdns.o
CC build/mdns/mdns_console.o
AR build/mdns/libmdns.a
CC build/micro-ecc/micro-ecc/uECC.o
AR build/micro-ecc/libmicro-ecc.a
/home/neoha/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/component_wrapper.mk:205: warning: overriding recipe for target 'libmicropython.a'
/home/neoha/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkrules.mk:132: warning: ignoring old recipe for target 'libmicropython.a'
Generating C:/esp32_win32_msys2_environment_and_toolchain-20180110/msys32/home/neoha/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/mpversion.h
GENlast /home/neoha/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last
/bin/sh: /home/neoha/MicroPython_ESP32_psRAM_LoBo/Tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Argument list too long
make[1]: *** [/home/neoha/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkrules.mk:76: /home/neoha/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last] Error 126
make[1]: *** Deleting file '/home/neoha/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last'
make: *** [/home/neoha/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/project.mk:451: component-micropython-build] Error 2
'make all' FAILED!
```

loboris, 2018-02-05:
I've just tried the instalation and building on fresh Windows 10 installation following the [instructions from Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/winsetup) and there were no problems.

loboris, 2018-02-12:
I'll close this issue.
Please, reopen if needed.

closed 2018-02-12 by loboris