loboris/MicroPython_ESP32_psRAM_LoBo #56: Problem with FTP server and lftp

Liberasys, 2018-02-06:
Hello Boris,

I still have some problems with lftp.
First, when TLS feature is negociated, the FTP server crashes:

> Type "help()" for more information.
> >>> W (16225) [Ftp]: FTP RESET
> W (16228) [Ftp]: Error sending command reply.
> ets Jun  8 2016 00:22:57

Secondly, lftp mirrorring is still not functionnal. I join you the lftp script I made and the wireshark capture. I think that the NLST command is not managed by FTP server.

```
#!/bin/bash
HOST='ftp://192.168.205.101'
USER='micro'
PASS='python'
TARGETFOLDER='/flash/'
SOURCEFOLDER='./filesystem/'

lftp -f "
set ftp:use-feat no
set ftp:ssl-allow no
set ftp:passive-mode yes
open $HOST
user $USER $PASS
lcd $SOURCEFOLDER
mirror --scan-all-first --reverse --delete --verbose --parallel=1 $SOURCEFOLDER $TARGETFOLDER
bye
"
```

```
220 Micropython FTP Server
USER micro
331 
PASS python
230 
PWD
257 /
CWD /flash
250 
CWD www
250 
PASV
227 (192,168,205,101,7,232)
LIST
150 
226 
TYPE I
200 
MDTM connecttest.txt
550 
MDTM get_config.pyhtml
550 
MDTM index.html
550 
MDTM ncsi.txt
550 
MDTM reset.pyhtml
550 
MDTM sys_info.pyhtml
550 
MDTM test2.pyhtml
550 
CWD
250 
MKD /flash
250 
MKD /flash/
250 
CWD /flash
250 
TYPE A
200 
PASV
227 (192,168,205,101,7,232)
LIST
150 
226 
PASV
227 (192,168,205,101,7,232)
NLST
502 
QUIT
221 
```

Anyway file transfert with filezilla is OK.
But an automatic sync with lftp would be perfect to me :-)
For information I found lemariva ftp server code which works well with lftp : https://github.com/lemariva/uPyPortal/blob/master/ftp.py
Maybe you could find some usefull information in it ?

loboris, 2018-02-06:
I have added `NLST` command to FTP server and a few other changes.
**lftp**. `mirror` command now works, I have to test it more and will commit when ready.

Liberasys, 2018-02-07:
Great !
After that, do you want that I provide the bash script with a small help in order to integrate it in your repository and/or wiki ?
I must say that ntp_sync + time properties in filesystem + FTP mirror based on files dates will be a must :-)

loboris, 2018-02-07:
The ftp server is now updated.
Not fully tested with **lftp** `mirror` command, so you should be carefull...

oldfartradioman, 2018-02-08:
Tested with Ftp-Remote package on Atom editor seems to be working fine.

Liberasys, 2018-02-09:
For information, AUTH TLS leads to problems (known issue). Some information:
network capture:

> 220 Micropython FTP Server
FEAT
502 no-features
AUTH TLS
332 
...........Z}w|.y.......@..p..e....4
........r.,.......
.$.s.+.....	.#.r...0.......(.w./.....'.v.....{...5.=.......z.../.<.A...
...}.....9.k.......|...3.g.E.......b..........................192.168.205.101......#...
...
.................
........................332 

Firmware problem:

> >>> W (21757) [Ftp]: FTP RESET
W (21760) [Ftp]: Error sending command reply.

Liberasys, 2018-02-09:
I confirm it is OK with lftp now ! Thank you Boris !!! :-)
If it can help, you will find here under my small script for syncing local "./filesystem" folder to "/flash/" folder on ESP32. It is very practical for developing.

First you have to do this with ESP32:
- connect network
- run loboris FTP server
- sync with NTP

In the script, change the IP of your ESP32, the source folder and you are good for basic setup.

`#!/bin/bash
HOST='ftp://192.168.205.101'
USER='micro'
PASS='python'
TARGETFOLDER='/flash/'
SOURCEFOLDER='./filesystem/'

#set ftp:use-feat no
lftp -f "
set ftp:passive-mode yes
set ftp:ssl-allow no
open ${HOST}
user ${USER} ${PASS}
lcd ${SOURCEFOLDER}
mirror --scan-all-first --reverse --delete --verbose --exclude config.json --parallel=1 ./ ${TARGETFOLDER}
bye
"
`

loboris, 2018-02-09:
I have added the negative response to `AUTH` command
If the client tries to establish the secure connection sending `AUTH TLS` command, the `AUTH TLS` will be rejected by the server and **lftp** will continue with unsecure connection.
So, there is no need any more to set `set ftp:ssl-allow no`.

```
boris@UbuntuMate:~$ lftp -d
lftp :~> open 192.168.0.16
---- Resolving host address...
---- 1 address found: 192.168.0.16
lftp 192.168.0.16:~> user micro python
lftp micro@192.168.0.16:~> ls
---- Connecting to 192.168.0.16 (192.168.0.16) port 21
<--- 220 Micropython FTP Server
---> FEAT
<--- 502 no-features
---> AUTH TLS
<--- 504 not-supported
---> USER micro
<--- 331 
---> PASS python
<--- 230 
---> PWD
<--- 257 /
---> PASV
<--- 227 (192,168,0,16,7,232)
---- Connecting data socket to (192.168.0.16) port 2024
---- Data connection established
---> LIST
<--- 150 
drw-rw-rw-   1 root  root         0 Feb 09 2018 flash
---- Got EOF on data connection
---- Closing data socket
<--- 226 
lftp micro@192.168.0.16:~> 
```

This change will be committed on the next update.

Liberasys, 2018-02-09:
Perfect, thank you Boris !
I'm happy to help a bit in return :-)

Liberasys, 2018-02-09:
Just a "quick start" example in more, for testers in the hurry or beginners (wahooo effect :-) : 

```
# connect to a WIFI AP, sync time and start telnet and FTP server
# Paste it in the Python command line (REPL)

wifi_ssid = "my_home_ssid"
wifi_passwd = "my_wifi_password"
my_timezone = "CET-1CEST" # found in second field, text before the coma, in https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/docs/zones.csv

import network
import machine
import time

sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
sta_if.connect(wifi_ssid, wifi_passwd)

time.sleep(6)
sta_if.ifconfig()

rtc = machine.RTC()
rtc.init((2018, 01, 01, 12, 12, 12))
rtc.ntp_sync(server= "", tz=my_timezone, update_period=3600)
network.ftp.start(user="micro", password="python", buffsize=1024, timeout=300)
network.telnet.start(user="micro", password="python", timeout=300)

```

closed 2018-03-06 by loboris