loboris/MicroPython_ESP32_psRAM_LoBo #57: Idea of improvment : ring buffer syslogging

Liberasys, 2018-02-07:
Hello,

I don't think it is already implemented.
A nice feature could be a "syslog like" thread that receives logs of system and programs.
It would store logs in a configurable size ring buffer (with log date/time, log source, log level, log entry as text).
Verbosity could be set per item at producer level (program/system) or by filter at syslog level (consumer).
A module could be written in order to get/flush/filter/continuously display log entries.

This way, no need to write to UART. And user would be able to get the logs when he wants.
It may be easier for debugging purposes.

Just an idea. What do you think about Boris ?

loboris, 2018-02-07:
There is the basic support for esp-idf logging redirection. It will be exposed in some future update, so it will be possible to redirect it to some (ring)buffer or even file.
The problem is that it can contain quite large amount of data and can happen quite often if enabled, so the memory (or file) space and performance impact is an issue.

Liberasys, 2018-02-09:
> There is the basic support for esp-idf logging redirection. It will be exposed in some future update, so it will be possible to redirect it to some (ring)buffer or even file.

Nice :-)

> The problem is that it can contain quite large amount of data and can happen quite often if enabled, so the memory (or file) space and performance impact is an issue.

Yes, of course. The basic configuration should not be verbose. And the user shall be warn of potential performance issue caused by increasing the log level. Anyway, with a non verbose system, a 10kB ring buffer in memory could be sufficient. With logs like WIFI connected, thread started, FTP client connected.

closed 2018-02-09 by Liberasys