loboris/MicroPython_ESP32_psRAM_LoBo #58: Idea of improvment : init system

Liberasys, 2018-02-07:
The idea is to have a sort of conventions that will permit to start automatically programs and threads at boot.

For example:
/flash/init/10p_configure_network.py
/flash/init/20p_start_web_server.py
/flash/init/30p_start_dns_server.py
/flash/init/40t_my_program_as_thread_1.py
/flash/init/50t_my_program_as_thread_2.py

All the files in /flash/init will be taken in account, by numerical order.
Two first characters of the filename defines priority.
Third character is "p" for program or "t for thread".
Then the name of the program/thread.
Than ".py"

In all those files, a simple method (def pynit_start() for example) will be called (programm) or started as separate thread.

We can then imagine other methods : pynit_stop, pynit_suspend, pynit_restart, ...

It may seems to be overkill but I had implemented that in LUA with ESP8266 and it helped me to make projects faster and with re-usability...

What do you think about Boris ?

loboris, 2018-02-07:
You can implement that in your `boot.py` or `main.py` easily.

I'll consider that idea for the future development.

Liberasys, 2018-02-09:
Yes of course, I can implement it in Python.
For the moment it is not planned.
Glad to see that this idea has some echo :-)

closed 2018-02-09 by Liberasys