loboris/MicroPython_ESP32_psRAM_LoBo #59: Question : how to share variables / configuration / data between codes/threads ?

Liberasys, 2018-02-07:
In my project, I store some configuration in a json file.
This json file is parsed at boot in order to get network configuration.
This json file is parsed and written in a method called from the JC webserver in order to display and reconfigure network connection for example.
I did that because I had not seen any simple way to share information between init codes and threads.
Did I miss something ?
Would it be interesting to have a dedicated thread code that read, store in RAM, store in file some data with external methods in order to access those data ?

kevinkk525, 2018-02-07:
I don't see where this questions are specific to the loboris-port and they are definitely not a bug report. Please ask questions in the forum so issues and questions are still being separate, except if @loboris has a different point on view on this, I don't know how he handles this kind of issue report.

Anyways, I'll shortly answer the question: I parse a json file as well but store the configs in a dictionary in a module that every thread/module imports. you just have to make sure you don't change the dictionary at the same time from different threads to be safe (or use locks). Problem solved (except if I did not understand what you meant).
To your last question: what kind of code should that be? writing to a file on every change is a bad idea and you could easily write a module like that yourself.

loboris, 2018-02-07:
Unfortunately, there is no place on GitHub for such a suggestions and discussions.
I agree with @kevinkk525 that the MicroPython forum should be more appropriate place for such a discussion and more people could see it and express their opinion.

There are many ways of sharing the data, global variables, NVS or RTC memory variables (if the data has to be preserved during power off/deepsleep), inter-thread messaging and notifications, etc. It all depends on your application and should be quite easy to implemented.

Liberasys, 2018-02-09:
@kevinkk525 
Honestly, I was asking myself if it was the right place to ask...
I am OK with the json module.
Code that in fact will be a small BDD server like. But in fact, after some reflexion, what I want is something like sqlite. Thus my need is already satisfied. I would prefer something more integrated and more simple but we won't reinvent the weel after all :-)
I will share my json module code. Maybe somebody will be interested in improving it with some semaphore/locking system :-)

@loboris 
I agree.

Sorry for inconvenience guys.

closed 2018-02-09 by Liberasys