loboris/MicroPython_ESP32_psRAM_LoBo #60: High latency UART

wootsrx, 2018-02-08:
**Dear Loboris**

**Sorry for my poor English.**

**Thank you very much for your incredible project.**
**I did my project with official Micropython but I had trouble with Wifi (STA mode and socket server), it could not work after AP turned off then turned on. But your firmware is pretty good about that point.**

**In my project, I send command via UART to a machine then it will response back.
With official Micropython, I can use uart.write() command then wait only 0.05 sec i can get response by uart.read()**
**My code and response are below**

MicroPython v1.9.2-275-gd0e11f7-dirty on 2018-01-24; ESP32 module with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import UART
=== import time
=== uart=UART(2,19200,tx=17,rx=16)
=== uart.init(19200,bits=8,parity=0,stop=1)
=== text = [4,64,81,5]
=== ready = bytearray(text)
=== uart.write(ready)
=== time.sleep(.05)
=== DATA = uart.read()
=== print(DATA)
I (134941) uart: ALREADY NULL
I (134941) uart: queue free spaces: 10
4
b'\x04'
>>>

**By your firmware I cannot get response with wait 0.05sec**

MicroPython ESP32_LoBo_v3.1.17 - 2017-02-05 on ESP32 board with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import UART
=== import time
=== uart=UART(2,19200,tx=17,rx=16)
=== uart.init(19200,bits=8,parity=0,stop=1)
=== text = [4,64,81,5]
=== ready = bytearray(text)
=== uart.write(ready)
=== time.sleep(.05)
=== DATA = uart.read()
=== print(DATA)
=== uart.flush()
4
b''
>>>

**I can get response with wait 1 sec**

MicroPython ESP32_LoBo_v3.1.17 - 2017-02-05 on ESP32 board with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import UART
=== import time
=== uart=UART(2,19200,tx=17,rx=16)
=== uart.init(19200,bits=8,parity=0,stop=1)
=== text = [4,64,81,5]
=== ready = bytearray(text)
=== uart.write(ready)
=== time.sleep(1)
=== DATA = uart.read()
=== print(DATA)
=== uart.flush()
4
b'\x04'

**I also try by wait 0.1 sec, 0.5 sec not good.**
**Could you please give me suggestion how to improve the response time?**
**Thank you in advance.**

**Woot**

loboris, 2018-02-08:
The received character should be available with far lower latency (as soon as it is received).
I'll check what could cause a delay.
Thanks for reporting.

loboris, 2018-02-08:
To test this issue, I've created a simple Python function on PC which waits for 4 bytes on serial port and sends 4 byte response:
```
import serial
ser.baudrate = 19200

def test():
    res = ser.read(4)
    ser.write(res)
    print(res)

test()
```
Then on MicroPython:
```
from machine import UART
import time
uart=UART(2,19200,tx=17,rx=16)
uart.init(19200,bits=8,parity=0,stop=1)
```
Executed:
```
>>> uart.write("1234"); time.sleep(0.05); uart.read(4);
4
b'1234'
>>> 
```
The response was available after 0.05 sec.

Then, to test the response time:
```
uart.init(timeout=100) # set uart timeout
>>> start=time.ticks_us(); uart.write("1234"); uart.read(4); end=time.ticks_us()
4
b'1234'
>>> time.ticks_diff(end,start)
10675
>>> 
```
The response was available after ~10 ms.
(The response time is acctually shorter, but when `uart.read()` waits for the response, the input buffer is checked in 10 ms intervals. This could be lowered, I'll test it)

**Conclusion:**
The explanation for your issue could be that you device acctually does not respond after 50 ms, but after some larger interval, and the **official Micropython** `uart.read()` acctualy introduces a long delay, so that `time.sleep(0.05)` has almost no effect.

wootsrx, 2018-02-09:
Dear Loboris

Thank you very much for your suggestion and explanation, i will test accordingly then i will report you the result.

Best regards,

Woot

wootsrx, 2018-02-16:
Hi Boris

Sorry for my late response.
I try by loop Tx and RX (pin 16  and 17) then test by the following code

from machine import UART
import time,gc
uart=UART(2,19200,tx=17,rx=16)
uart.init(19200,bits=8,parity=0,stop=1)
uart.flush()
for i in range(5): 
    gc.collect()
    data = "0000"+str(i)
    DATA = bytes(data,"utf-8")
    start=time.ticks_us(); uart.write(DATA); time.sleep(0.05);uart.read(5); end=time.ticks_us()
    print(time.ticks_diff(end,start))
    uart.flush()

Then i got 
5
b'00000'
52108
5
b'00001'
51734
5
b'00002'
50903
5
b'00003'
52880
5
b'00004'
50903

**so I understand, it was not UART latency** 

**PS. If i try run the 2nd time of my code, it can not get data from uart.read() and found the error then CPU halted.**
**Could you please suggest me what is the cause of the error?**
**I run with nano32 (esp32 with 4MB flash, no psRAM )**

FreeRTOS running on BOTH CORES, MicroPython task started on App Core.
Running from Factory partition starting at 0x10000, [MicroPython].

 Reset reason: RTC WDT reset
    uPY stack: 19456 bytes
     uPY heap: 80000/5856/74144 bytes

MicroPython ESP32_LoBo_v3.1.17 - 2017-02-05 on ESP32 board with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import UART
=== import time,gc
=== uart=UART(2,19200,tx=17,rx=16)
=== uart.init(19200,bits=8,parity=0,stop=1)
=== uart.flush()
=== for i in range(5):
===     gc.collect()
===     data = "0000"+str(i)
===     DATA = bytes(data,"utf-8")
===     start=time.ticks_us(); uart.write(DATA); time.sleep(0.05);uart.read(5); end=time.ticks_us()
===     print(time.ticks_diff(end,start))
===     uart.flush()
5
b'00000'
52108
5
b'00001'
51734
5
b'00002'
50903
5
b'00003'
52880
5
b'00004'
50903
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import UART
=== import time,gc
=== uart=UART(2,19200,tx=17,rx=16)
=== uart.init(19200,bits=8,parity=0,stop=1)
=== uart.flush()
=== for i in range(5):
===     gc.collect()
===     data = "0000"+str(i)
===     DATA = bytes(data,"utf-8")
===     start=time.ticks_us(); uart.write(DATA); time.sleep(0.05);uart.read(5); end=time.ticks_us()
===     print(time.ticks_diff(end,start))
===     uart.flush()
5
b''
51627
5
b''
51795
5
b''
50931
5
b''
50910
5
b''
50932
>>> assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "C:/msys32/home/User/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/portmux_impl.h", line 105, function: vPortCPUAcquireMutexIntsDisabled
abort() was called at PC 0x4013adf3 on core 0

Backtrace: 0x4008b4ec:0x3ffd8ba0 0x4008b68f:0x3ffd8bc0 0x4013adf3:0x3ffd8be0 0x40087a73:0x3ffd8c10 0x40089664:0x3ffd8c30 0x40134a82:0x3ffd8c70

CPU halted.

loboris, 2018-02-16:
When you paste your code sequence the 2nd time a new uart instance is created on the same hardware UART.
It looks like there is a bug in reinitializing the hardware uart which causes the crash.
I'll try to fix it as soon as possible.

If you remove the lines
```
=== uart=UART(2,19200,tx=17,rx=16)
=== uart.init(19200,bits=8,parity=0,stop=1)
```
from your 2nd run, it should work.

wootsrx, 2018-02-16:
Thank you so much for your kindly help.
Could you please comment about hardware? My device nano32 ,is it suitable with your firmware?
[https://www.cytron.io/p-nano32](url)
or I should use wrover module.

Thank you for your help.

loboris, 2018-02-16:
nano32 should work fine (any ESP32 board can be used).

Using the board with psRAM (spiram) is recommended as you will have far more heap space.
TTGO T8 V1.1 (links on wiki), for example, has 4MB psRAM, sdcard slot, battery charger, for half the price of nano32 ...

wootsrx, 2018-02-16:
Thank you very much for your suggestion.
Yes, I will try on hardware that you recommenced.

Thank you and happy Chinese new year.

closed 2018-02-16 by wootsrx