loboris/MicroPython_ESP32_psRAM_LoBo #61: your distribution doesn't build esp32-idf template project

mfp20, 2018-02-08:
It ends with the following error:

```
/esp-idf-with-micropython_psRAM_by_LoBo/Tools/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:11:23: fatal error: mphalport.h: No such file or directory
compilation terminated.
```
I'm using 

```
commit d775bb11a0341fbf4da8d9801fc99adcc54393f5 (HEAD -> master, origin/master, origin/HEAD)
Author: Boris Lovosevic <loboris@gmail.com>
Date:   Wed Feb 7 17:51:08 2018 +0100
```
Regards

loboris, 2018-02-09:
The included esp-idf is intended only for for compiling this project.

It includes the patches to the following components:
`spiffs` - enable directories in spiffs, make format faster when some blocks are already erased
`lwip/apps/sntp` - enable end of synctonization detection from the main program
`drivers/spi-master` - enable the access to the low level spi-master functions and usage of more than 3 CS on the same bus
You can see the diff [here](http://loboris.eu/ESP32/esp-idf.diff)

Your error is caused by the patch in `components/spiffs/spiffs/src/spiffs_hydrogen.c` which uses the function from MicroPython's `mphalport` resetting the watchdog during the format.

mfp20, 2018-02-09:
Thanks Boris.

Are those patches needed by micropython? The more we move away from espressif, the less chances to merge with future mainstream updates.
To make you an example: I'd like to keep "arduino compatibility" by running one arduino xTask on core0, concurrently to micropython on core1, but if you keep shifting away from mainstream codebase, soon or later it won't be possible to keep the benefits of both worlds! I've spent the last 3 hours trying to merge the arduino esp32 core into your project but I couldn't (yet), because I couldn't find a way to add options to Kconfig (menuconfig). 

Any advice?

loboris, 2018-02-09:
I don't think this is "moving away from espressif". I have no intention of changing anything in esp-idf.
The `sntp` patch is a realy small one, not realy needed, but convenient.
The `spiffs` patch is the essential one if we want to have the directories support on spiffs. I've issued a PR in esp-idf repository, but it is not merged (yet).
The `spi-master` patch is important for the way I'm using the spi driver. The PR was also issued.

I don't think there will be any other pathches to the esp-idf.

I don't think mixing arduino libraries with MicroPython is a good idea, but it shouldn't be too difficult.
If you need to add some config options, do it in `MicroPython_BUILD/components/micropython/Kconfig.projbuild`

mfp20, 2018-02-09:
Well, I didn't say that you want to move away from espressif (or anything else). Usually it just happens ... by not caring of consequences. Have a look at my other issue I posted a few minutes ago: I'd like to enable QIO for my board but I don't know where to do it: your repo, espressif's one or ... just mine? And what about arduino-esp32 one? Having all those features in one codebase boosts development: everybody can take advantage of each and every improvement.

Just for curiosity sake: how did you issue PRs back to esp-idf without having a direct relationship between your repo and esp-idf one? I mean, I can't see the 'fork' link under your repo's name.

Can you please elaborate a bit why "isn't a good idea"? I've spotted a few users on my board's forum willing to go back to arduino-only firmware because they can't manage micropython. By mixing the two worlds it would be easier for everybody. To make you an example, I'm doing it because I need some device libraries that already exist for arduino; without the arduino part I should write some python wrappers around some new C code.

Thanks for Kconfig.projbuild tip!!!!

loboris, 2018-02-09:
The **esp-idf** directory in my repository is just a **copy** of the [Espressif esp-idf](https://github.com/espressif/esp-idf)

The discussion about mixing arduino libraries with MicroPython could be a long one. Maybe a MicroPython forum would be a better place for it.

I have a [esp-idf fork](https://github.com/loboris/esp-idf) with some PR's (branches spiffs_new & spimaster)

mfp20, 2018-02-09:
I understand now. Thanks for making the thing more clear for me.

closed 2018-03-06 by loboris