loboris/MicroPython_ESP32_psRAM_LoBo #62: QIO mode is not enabled.

mfp20, 2018-02-09:
I checked the GigaDevice GD25Q32C datasheet: it supports QIO r/w ops. The board schema shows 4 data lines connected with mcu. But the bootloader doesn't enable the QIO mode.

In /components/bootloader/subproject/main/flash_qio_mode.c there's no entry for this chip (mfg ID C8, flash ID 4016), but it says that the default catch-all entry will manage it.
The point is that... well... I need to keep micropython in place but it's not worth to spend time solving issues on this codebase because it's not a fork from the main esp32-idf repo.

Again: is there a chance that you rebase your repo so that pull requests into espressif's esp32-idf one are possible?

loboris, 2018-02-09:
I don't realy understant what is the issue.
Absolutely nothing was ever changed in `components/bootloader/subproject/main/flash_qio_mode.c`.

If you want to make any PR to esp-idf, do it [here](https://github.com/espressif/esp-idf/pulls).

mfp20, 2018-02-09:
The issue is that ... I need micropython! If I raise the issue on esp-idf repo, it won't change in yours ...

loboris, 2018-02-09:
Yes, it will. Maybe with the delay of a couple of days, but it will.
I synchronize esp-idf to the master esp-idf repository mostly once a week.

The only reasons why I keep a copy of esp-idf repository in my repository are:
* couple of chages (patches) I find important for MicroPython
* somethimes, some changes in esp-idf breaks the MicroPython build. The esp-idf is updated only when I resolve all the issues
* to make the build process easier for more users

mfp20, 2018-02-09:
Ok, I posted this issue in esp-idf repo. Thanks.

closed 2018-03-06 by loboris