loboris/MicroPython_ESP32_psRAM_LoBo #63: DHT22 example ?

Liberasys, 2018-02-09:
Hello Boris,
Could you give an example for getting temperature and humidity from a DHT22 please ?
An "import DHT" gives an error and I cannot find appropriate methods :
`
>>> import machine
>>> d = machine.DHT(machine.Pin(26))
>>> d.
readinto        read            DHT11           DHT2X

`

BluSpanner, 2018-02-09:
You need to specify the type of DHT
for a DHT22 use
`d = machine.DHT(machine.Pin(26), machine.DHT.DHT2X)`

Sample code would be
```
import machine
pin = 26
dht = machine.DHT(machine.Pin(pin), machine.DHT.DHT2X)
result, temperature, humidity = dht.read()
if not result:
    print('Failed!')
else:
    print('t={} C'.format(temperature))
    print('h={} % RH'.format(int(humidity)))
```

   / David

Liberasys, 2018-02-09:
Great thank you David, works perfectly ! :-)

loboris, 2018-02-09:
I'm (slowly) working on finishing the Wiki documentation for all the `machine` modules.
DHT documentation should be available soon.

Liberasys, 2018-02-09:
@loboris 
Thank you for the doc, it helps a lot.
@BluSpanner : how did you know how to do with the loboris firmware ?

hargathor, 2018-02-17:
@BluSpanner  Thanks a lot for the example but I'm curious on how you manage to find out how to do it. It coul help me on how to investigate for the next time while waiting on the documentation. Or even contributing to the doc!

BluSpanner, 2018-02-18:
Python's help() & dir() & TAB completion can be used to find out...
**Rather than document - here's how you can reverse-engineer...**
I understand that's what you want.
And I'm unable to edit the wiki

Let get started...

Start with `help()` in the REPL

```
>>> help()
Welcome to LoBo MicroPython on the ESP32!

For online documentation please visit:
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki

For access to the hardware use the 'machine' module:

import machine
pin12 = machine.Pin(12, machine.Pin.OUT)
pin12.value(1)
pin13 = machine.Pin(13, machine.Pin.IN, machine.Pin.PULL_UP)
print(pin13.value())
i2c = machine.I2C(scl=machine.Pin(21), sda=machine.Pin(22))
i2c.scan()
i2c.writeto(addr, b'1234')
i2c.readfrom(addr, 4)

Basic WiFi configuration:

import network
sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
sta_if.scan()                             # Scan for available access points
sta_if.connect("<AP_name>", "<password>") # Connect to an AP
sta_if.isconnected()                      # Check for successful connection

Control commands:
  CTRL-A        -- on a blank line, enter raw REPL mode
  CTRL-B        -- on a blank line, enter normal REPL mode
  CTRL-C        -- interrupt a running program
  CTRL-D        -- on a blank line, do a soft reset of the board
  CTRL-E        -- on a blank line, enter paste mode

For further help on a specific object, type help(obj)
For a list of available modules, type help('modules')
```

Notice the `help('modules')` on the last line.

Let use it to show a list of modules have been frozen into the firmware
`help('modules')`

```code
>>> help("modules")
__main__          json              ssl               upysh
_thread           logging           struct            urandom
array             machine           sys               ure
binascii          math              time              urequests
builtins          microWebSocket    tpcalib           uselect
cmath             microWebSrv       ubinascii         usocket
collections       microWebTemplate  ucollections      ussl
display           micropython       uctypes           ustruct
errno             network           uerrno            utime
framebuf          os                uhashlib          utimeq
freesans20        pye               uheapq            uzlib
functools         random            uio               writer
gc                re                ujson             ymodem
hashlib           select            uos               zlib
heapq             socket            upip
io                ssd1306           upip_utarfile
[... 154 lines cut ...]
>>> dht.read()
(True, 27.0, 47.0)
```

---

Expanded

```
import machine
p4 = machine.Pin(4)
dht_type = machine.DHT.DHT2X
dht = machine.DHT(p4, dht_type)
(success, temperature, humidity) = dht.read()
```

---
so altogether, with a shortened version, using positional arguments

```
>>> import machine
>>> machine.DHT(machine.Pin(4), machine.DHT.DHT2X).read()
(True, 26.0, 44.0)
>>> 
```

hargathor, 2018-02-19:
@BluSpanner : thanks a lot for this very complete walkthrough!  That's very usefull

Liberasys, 2018-03-09:
Sorry for late response.
@BluSpanner : thank you for this detailed process, it will help !

closed 2018-03-09 by Liberasys