loboris/MicroPython_ESP32_psRAM_LoBo #64: Build menuconfig script fails

jimwims, 2018-02-09:
I am trying to update to the latest version following the instructions now included in the wiki.  I have done this previously with no problems.  This time after I run ' git pull',  when I try to run './BUILD.sh menuconfig', I get the following:

      MENUCONFIG
      'make menuconfig' FAILED!

Where have I gone wrong here?

Jim

jimwims, 2018-02-09:
I am using macOS 10.13.3 with an ESP-WROVER-KIT v3.

Jim

loboris, 2018-02-09:
I have tried it on my MacOS High Sierra in VirtualBox (MicroPython not updated for more than a month).
`git pull` -> `./BUILD.sh menuconfig` -> `./BUILD.sh -j8 clean all` and there were no errors.

You can try to delete `MicroPython_ESP32_psRAM_LoBo/Tools/ver20xxxxxx.id` and then run `./BUILD.sh menuconfig`, the esp-idf and toolchain will be deleted and unpacked and menuconfig should run.

If nothing helps, try to make a fresh clone...

jimwims, 2018-02-10:
I was trying to do the build in a terminal window in Atom.  Through a rather circuitous process, I learned that I had not allowed for enough lines in the terminal window.  I went back to using the standard Apple 20 x 80 terminal and everything seems to be working fine.

Thanks for your help.

Is there way to make a donation to you other than through PayPal?  I  have had bad experiences with PP in the past so I avoid using it.

loboris, 2018-02-10:
I'm glad it works.

Sorry, only PayPal. Just out of curiosity, what kind of problems did you have? I'm using it for years with no issues.

closed 2018-02-12 by loboris