loboris/MicroPython_ESP32_psRAM_LoBo #67: MicroWebSrv not started when in AP mode

Liberasys, 2018-02-12:
Hello Boris,
You added a check of WLAN connected before running 
>     def Start(self, threaded=True) :
        if not self._started :
            if not network.WLAN().isconnected():
                print("WLAN not connected!")
                return

But when the ESP32 is in AP mode, MicroWebSrv doesn't start neither...

Maybe you could follow the example of lemariva in his FTP server ? It checks the two cases :
```

        self.wlan = network.WLAN(network.AP_IF)
        if self.wlan.active():
            ifconfig = self.wlan.ifconfig()
        else:
            self.wlan = network.WLAN(network.STA_IF)
            if self.wlan.active():
                ifconfig = self.wlan.ifconfig()
            else:
                dbg("No active connection")
```

loboris, 2018-02-12:
Thank you for informing about that.
This check should be changed to check for AP mode too, or removed (it is simple to check the connection before starting the server).

elektrobil, 2018-03-14:
Hi again Boris
I have same issue did yo solve this problem ? I have the latest ver. I have Two connection AP mode and LAN8720.Lan module connect ntp server and sync the time but web server not working I can not reach the server both of them
Ps. : Wifi STA mode works good.
Thank you so much

loboris, 2018-03-18:
I will update microWebSrv soon (probably next week). Then I'll include also the AP fix.

loboris, 2018-04-12:
microWebServer was updated, it now works both in STA and AP modes.

closed 2018-04-12 by loboris