loboris/MicroPython_ESP32_psRAM_LoBo #68: FTP SERVER stack overflow  Error

elektrobil, 2018-02-12:
Hi Boris,
When I connect via filezilla ftp in ESP32 first time I could wrote files but second time I can connect the ESP32 but I cant get in the directories ESP32 make an errror like this
```

>>> ***ERROR*** A stack overflow in task FtpServer has been detected.
abort() was called at PC 0x40090a68 on core 1
Backtrace: 0x40090854:0x3ffe8b00 0x40090a4f:0x3ffe8b20 0x40090a68:0x3ffe8b40 0x4008d1f6:0x3ffe8b60 0x4008f008:0x3ffe8b80 0x4008efbe:0x00015180
```

loboris, 2018-02-12:
I'm using ftp server a lot and never had such an issue.
Ftp server task runs with 6K of stack and maximum stack space it uses is ~4.5K

If you are running on ESP32 without psRAM, try to lower the MicroPython heap size in menuconfig to some lower value , e.g. 72K (`→ MicroPython → System settings → MicroPython heap size (KB)`) and rebuild the firmware (`./BUILD.sh clean all`)

elektrobil, 2018-02-13:
Im using psRAM 4MB **problem solved** Im changing FileZilla setings like Wiki page 
Thank you so much

> Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 2621440; Flash address: 0x180000]
>
> Filesystem size: 2402304 B
>            Used: 3584 B
>            Free: 2398720 B
> 
> 
> FreeRTOS running on BOTH CORES, MicroPython task running on both cores.
> Running from Factory partition starting at 0x10000, [MicroPython].
> 
>  Reset reason: RTC WDT reset
>     uPY stack: 64512 bytes
>      uPY heap: 3073664/5664/3068000 bytes (in SPIRAM using malloc)
> 
> MicroPython ESP32_LoBo_v3.1.18 - 2017-02-07 on BATTLOG with ESP32
> Type "help()" for more information.

closed 2018-02-13 by elektrobil