loboris/MicroPython_ESP32_psRAM_LoBo #69: Guru Meditation Error On ESP32

elektrobil, 2018-02-13:
Hi Boris,

I need your help again ESP32 make an ERROR 

> Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
> .Exception was unhandled.
> Core 0 register dump:
> PC      : 0x4012f3b6  PS      : 0x00060c30  A0      : 0x8013c22e  A1      : 0x3ffeed30
> A2      : 0x00000000  A3      : 0x3f819ad0  A4      : 0x3f819a80  A5      : 0xfffffffb
> A6      : 0x00000000  A7      : 0x4012f2ac  A8      : 0x8012f3b6  A9      : 0x3ffeed10
> A10     : 0x83126e97  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3f818dac
> A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x83126eaf  LBEG    : 0x4009f1d5  LEND    : 0x4009f1e5  LCOUNT  : 0xfffffffe
> 
> Backtrace: 0x4012f3b6:0x3ffeed30 0x4013c22b:0x3ffeed70 0x40132b96:0x3ffeee10 0x4012e3b9:0x3ffeee40 0x4013ae68:0x3ffeee60 0x401152b5:0x3ffeeef0)

I made a coredump file and compile the output like that. What can I do for that. Can you help me about this problem. ESP32 makes this error running after about ~30-50 mins 

> ==================== ESP32 CORE DUMP START ====================
> 
> ================== CURRENT THREAD REGISTERS ===================
> /builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: value_of_register_lazy: Assertion frame_id_p (get_frame_id (frame)) failed.
> A problem internal to GDB has been detected,
> further debugging may prove unreliable.
> Quit this debugging session? (y or n) [answered Y; input not from terminal]
> /builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: value_of_register_lazy: Assertion `frame_id_p (get_frame_id (frame)) failed.
> A problem internal to GDB has been detected,
> further debugging may prove unreliable.
> Create a core file of GDB? (y or n) [answered Y; input not from terminal]
> GDB exited (None / None)!
> Problem occured! GDB exited, restart it.
> Reading symbols from MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esptool_py/esptool/test/elf2image/esp32-app-template.elf...done.
> [New <main task>]
> [New process 1]
> [New process 2]
> [New process 3]
> [New process 4]
> [New process 5]
> [New process 6]
> [New process 7]
> [New process 8]
> [New process 9]
> [New process 10]
> [New process 11]
> [New process 12]
> [New process 13]
> [New process 14]
> [New process 15]
> [Current thread is 1 (<main task>)]
> 
> ==================== CURRENT THREAD STACK =====================
> #-1 0x4012f3b6 in ?? ()
> Backtrace stopped: Cannot access memory at address 0x4012f3b6
> 
> ======================== THREADS INFO =========================
>   Id   Target Id         Frame 
>   16   process 15        0x4008ca79 in rtc_deep_slp_conf ()
>   15   process 14        0x4008ca79 in rtc_deep_slp_conf ()
>   14   process 13        0x4008ca79 in rtc_deep_slp_conf ()
>   13   process 12        0x4008ca79 in rtc_deep_slp_conf ()
>   12   process 11        0x4008ca79 in rtc_deep_slp_conf ()
>   11   process 10        0x4008ca79 in rtc_deep_slp_conf ()
>   10   process 9         0x4008ca79 in rtc_deep_slp_conf ()
>   9    process 8         0x40081976 in xQueueGiveFromISR (xQueue=0x3ffc7fe8, pxHigherPriorityTaskWoken=0x3ffc8030) at /home/gus/esp/32/idf/components/freertos/./queue.c:1235
>   8    process 7         0x4008ca79 in rtc_deep_slp_conf ()
>   7    process 6         0x4008dcee in rtc_digital_lp_mode_off_stg1 ()
>   6    process 5         0x4008ee3e in ?? ()
>   5    process 4         0x4008ca79 in rtc_deep_slp_conf ()
>   4    process 3         0x4008ca79 in rtc_deep_slp_conf ()
>   3    process 2         __truncdfsf2 () at /home/ivan/e/crosstool-NG/.build/src/gcc-4.8.5/libgcc/config/xtensa/ieee754-df.S:2307
>   2    process 1         __truncdfsf2 () at /home/ivan/e/crosstool-NG/.build/src/gcc-4.8.5/libgcc/config/xtensa/ieee754-df.S:2307
> * 1    <main task>       0x4012f3b6 in ?? ()
> 
> ======================= ALL MEMORY REGIONS ========================
> `Name   Address   Size   Attrs
> .iram0.vectors 0x40080000 0x400 R XA
> .iram0.text 0x40080400 0x16384 R XA
> .dram0.data 0x3ffbef70 0x12c0 RW A
> .flash.rodata 0x3f400010 0x50cc RW A
> .flash.text 0x400d0018 0x28958 R XA
> .rtc.text 0x400c0000 0x0 RW  
> .coredump.tasks 0x3ffeb200 0x17c RW 
> .coredump.tasks 0x3ffeec70 0x30c RW 
> .coredump.tasks 0x3ffd85a8 0x17c RW 
> .coredump.tasks 0x3ffd8440 0x160 RW 
> .coredump.tasks 0x3ffd8024 0x17c RW 
> .coredump.tasks 0x3ffd7eb0 0x16c RW 
> .coredump.tasks 0x3ffd781c 0x17c RW 
> .coredump.tasks 0x3ffd7630 0x1e4 RW 
> .coredump.tasks 0x3ffeaf18 0x17c RW 
> .coredump.tasks 0x3ffdc6b0 0x1ac RW 
> .coredump.tasks 0x3ffd8f2c 0x17c RW 
> .coredump.tasks 0x3ffd8da0 0x184 RW 
> .coredump.tasks 0x3ffdc2e0 0x17c RW 
> .coredump.tasks 0x3ffea7a0 0x178 RW 
> .coredump.tasks 0x3ffef494 0x17c RW 
> .coredump.tasks 0x3fff10d0 0x540 RW 
> .coredump.tasks 0x3ffd5f14 0x17c RW 
> .coredump.tasks 0x3ffd5d60 0x1ac RW 
> .coredump.tasks 0x3ffd6498 0x17c RW 
> .coredump.tasks 0x3ffd6310 0x180 RW 
> .coredump.tasks 0x3ffdc160 0x17c RW 
> .coredump.tasks 0x3ffdbfc0 0x198 RW 
> .coredump.tasks 0x3ffd90f4 0x17c RW 
> .coredump.tasks 0x3ffdadb0 0x1a0 RW 
> .coredump.tasks 0x3ffc35f0 0x17c RW 
> .coredump.tasks 0x3ffc3380 0x26c RW 
> .coredump.tasks 0x3ffd5958 0x17c RW 
> .coredump.tasks 0x3ffd57d0 0x180 RW 
> .coredump.tasks 0x3ffd9274 0x17c RW 
> .coredump.tasks 0x3ffdb7c0 0x194 RW 
> .coredump.tasks 0x3ffd799c 0x17c RW 
> .coredump.tasks 0x3ffd9d90 0x1bc RW 
> 
> Done!

loboris, 2018-02-13:
You can run esp-idf monitor as terminal, in case of an error you will get the complete error trace with the source file possition and the name of the function where the error occured.
`./BUILD.sh monitor`
Also, giving more detailed information about have you changed anything in the build and what is running  when the error occurs (any Python script) could help...

elektrobil, 2018-02-14:
Thank you for your Help again...

King Regards

closed 2018-02-14 by elektrobil