loboris/MicroPython_ESP32_psRAM_LoBo #70: CORRUPT HEAP

hargathor, 2018-02-14:
Hi Loboris. Once again I need your help. I'm running a mqtt subscriber/publisher script on the last version of your port of micropython and I get a corrupted heap error. Here is the stacktrace while running `./BUILD.sh monitor`

```
CORRUPT HEAP: multi_heap.c:370 detected at 0x3ffe3a50
abort() was called at PC 0x4008ad7b on core 0

Backtrace: 0x4008b4ec:0x3ffdbe90 0x4008b68f:0x3ffdbeb0 0x4008ad7b:0x3ffdbed0 0x40082f70:0x3ffdbef0 0x40082fa1:0x3ffdbf10 0x40083871:0x3ffdbf30 0x4000beaf:0x3ffdbf50 0x40087666:0x3ffdbf70 0x400878a0:0x3ffdbf90 0x400838da:0x3ffdbfb0 0x40083900:0x3ffdbfd0 0x40083a45:0x3ffdc000 0x40140deb:0x3ffdc020 0x4013c40d:0x3ffdc2e0 0x4013c395:0x3ffdc330 0x4008ab43:0x3ffdc360 0x4008ae3f:0x3ffdc380 0x4008305e:0x3ffdc3a0 0x4008387d:0x3ffdc3c0 0x4000bec7:0x3ffdc3e0 0x400f19cd:0x3ffdc400 0x400ea4fb:0x3ffdc420 0x400ec745:0x3ffdc450
0x40082f70: spi_flash_guard_start at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_ops.c:106
(inlined by) spi_flash_erase_range at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_ops.c:171

0x40082fa1: spi_flash_guard_end at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_ops.c:113
(inlined by) spi_flash_erase_range at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_ops.c:181

0x40083871: spi_flash_mmap_pages at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_mmap.c:241

0x40087666: rtc_clk_cpu_freq_value at /home/hargathor/dev/perso/esp-idf/components/soc/esp32/rtc_clk.c:388

0x400878a0: rtc_clk_apb_freq_get at /home/hargathor/dev/perso/esp-idf/components/soc/esp32/rtc_clk.c:593

0x400838da: spi_flash_mmap at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_mmap.c:115

0x40083900: spi_flash_munmap at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_mmap.c:254

0x40083a45: xt_set_interrupt_handler at /home/hargathor/dev/perso/esp-idf/components/freertos/./xtensa_intr.c:138

0x4008305e: spi_flash_guard_end at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_ops.c:113
(inlined by) spi_flash_write at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_ops.c:224

0x4008387d: spi_flash_mmap at /home/hargathor/dev/perso/esp-idf/components/spi_flash/./flash_mmap.c:106

CPU halted.
````

I think that's somehow linked to the mqtt subscriber but I don't know how. I tried lowering micropython Heap size in the menuconfig and also increasing the buffer payload size but no luck.
I'm using the Lolin32 Lite (from Wemos store) and I'm running this script as `main.py`
This code is not clean or finished so some functions are not called, other are deactivated or not impleted yet and some variables are not named properly (DEEP_SLEEP_TIME to not name it). But I think that is important to have a full view of the script running

```python
import _thread
import machine
import network
import ujson
import utime

global TOPIC

DEEP_SLEEP_TIME = 300000
DS18B20_PIN = 33
IDX = "77"
TOPIC = b'domoticz/in'
WIFI_MAX_TRIES = 20
working_threads = 0
LED_BUILT_IN = 22
# MOC actiing on positive alternance is on pin 19
ALT_POSITIVE_PIN = 19
# MOC actiing on negative alternance is on pin 0
ALT_NEGATIVE_PIN = 0

def default_mode():
    """TODO: Docstring for default_mode.
    :returns: TODO

    """
    confort()

def confort():
    """TODO: Docstring for confort.
    :returns: TODO
    Confort mode means no signal on the pilote wire
    """
[... 234 lines cut ...]
    print('Woke from a deep sleep')
else:
    print("Woke up from hard reboot ?")
pin = machine.Pin(LED_BUILT_IN, machine.Pin.OUT)
# Initialize the MOC3043 Pins
alt_positive_pin = machine.Pin(ALT_POSITIVE_PIN, machine.Pin.OUT)
alt_negative_pin = machine.Pin(ALT_NEGATIVE_PIN, machine.Pin.OUT)
default_mode()
# the temperature device is on GPIO16
dat = machine.Pin(DS18B20_PIN, machine.Pin.IN)
# create the onewire object

# ow = machine.Onewire(dat)
# ds = machine.Onewire.ds18x20(ow, 0)

# _thread.start_new_thread('th_led_flash', do_flashes, (5,))

# roms = init_onewire(ow)
nic = wifi_connect()
utime.sleep_ms(1000)
tempTimer = machine.Timer(2)
tempTimer.init(period=DEEP_SLEEP_TIME,
               mode=tempTimer.PERIODIC, callback=manage_temp)
pin.value(1)
````

kevinkk525, 2018-02-14:
I feal sorry for Boris if he has to read and understand all that code... Can you please reproduce your problem with a minimal code example?
If a function is not called, it is definitely not important. 
Also it would be nice to have a very small code that reproduces the issue as every hour Boris spends trying to reproduce an issue reported with a huge amount of code is time he does not have to develop this port.

hargathor, 2018-02-14:
Ok no problem I understand. I will clean up my code and come up with a smaller example

hargathor, 2018-02-15:
Here is an updated stacktrace with a more smaller example.

````
Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 2883584; Flash address: 0x140000]
----------------
Filesystem size: 2643200 B
           Used: 3584 B
           Free: 2639616 B
----------------
Starting One wire module with embeded temperature sensor
Woke up from hard reboot ?
Trying to connect to Wifi AP
Waiting for connection...
......('192.168.0.27', '255.255.255.0', '192.168.0.1', '192.168.0.35')
Connecting to MQTT broker
E (9638) [Mqtt client]: Socket error (128), exit Receive schedule
.[0®¤MU(] Disconnected
.[0®¤MU(] Connected
.[0®¤MU(] Connected
Subscribing to domoticz topic
CORRUPT HEAP: multi_heap.c:370 detected at 0x3ffe3b60
abort() was called at PC 0x4008ad7b on core 0
0x4008ad7b: multi_heap_assert at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/heap/./multi_heap.c:652
 (inlined by) multi_heap_malloc_impl at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/heap/./multi_heap.c:370

Backtrace: 0x4008b4ec:0x3ffe2500 0x4008b6c3:0x3ffe2520 0x4008ad7b:0x3ffe2540 0x40082f70:0x3ffe2560 0x40082fa1:0x3ffe2580 0x40083871:0x3ffe25a0 0x4000beaf:0x3ffe25c0 0x40137e05:0x3ffe25e0 0x4016dc0e:0x3ffe2610 0x4016dd8a:0x3ffe2660 0x4016deff:0x3ffe2680
0x4008b4ec: invoke_abort at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x4008b6c3: abort at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x4008ad7b: multi_heap_assert at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/heap/./multi_heap.c:652
 (inlined by) multi_heap_malloc_impl at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/heap/./multi_heap.c:370

0x40082f70: heap_caps_malloc at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/heap/./heap_caps.c:351

0x40082fa1: heap_caps_malloc_default at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/heap/./heap_caps.c:351

0x40083871: _malloc_r at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/newlib/./syscalls.c:37

0x40137e05: data_cb at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modmqtt.c:160

0x4016dc0e: deliver_publish at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/espmqtt/./mqtt.c:424

0x4016dd8a: mqtt_start_receive_schedule at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/espmqtt/./mqtt.c:513

0x4016deff: mqtt_task at /home/hargathor/dev/perso/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/espmqtt/./mqtt.c:608
````
```python
import _thread
import machine
import network
import ujson
import utime

DEEP_SLEEP_TIME = 300000
IDX = "77"
TOPIC = b'domoticz/in'
WIFI_MAX_TRIES = 20
LED_BUILT_IN = 22

def conncb(task):
    print("[{}] Connected".format(task))

def disconncb(task):
    print("[{}] Disconnected".format(task))

def subscb(task):
    print("[{}] Subscribed".format(task))

def pubcb(pub):
    print("[{}] Published: {}".format(pub[0], pub[1]))

def datacb(msg):
    data = ujson.loads(msg[2])
    if data.get("idx") == 78:
        print(int(data.get("svalue1")))

def wifi_connect():
    nic = network.WLAN(network.STA_IF)
    if not nic.isconnected():
        print("Trying to connect to Wifi AP")
        nic.active(True)
        nic.connect('SSID', 'WIFI-PASSWORD')
        nic.connect()
        print("Waiting for connection...")
        retries = 0
        while not nic.isconnected():
            if retries >= WIFI_MAX_TRIES:
                machine.reset()
            utime.sleep_ms(500)
            print('.', end='')
            retries += 1
    print(nic.ifconfig())
    utime.sleep(1)
    print("Connecting to MQTT broker")
    global client
    SERVER = '192.168.0.23'
    CLIENT_ID = machine.unique_id()
    client = network.mqtt(CLIENT_ID, SERVER,
                          cleansession=True, autoreconnect=True,
                          connected_cb=conncb, disconnected_cb=disconncb,
                          subscribed_cb=subscb, published_cb=pubcb,
                          data_cb=datacb)
    while client.status()[0] is not 1:
        utime.sleep_ms(500)
        print('.', end='')
    print("Subscribing to domoticz topic")
    client.subscribe('domoticz/out')
    return nic

def manage_temp(timer):
    pin.value(0)  # Pin is inverted for builtin
    print(nic.ifconfig())
    utime.sleep(5)
    pin.value(1)  # Pin is inverted for builtin

pin = machine.Pin(LED_BUILT_IN, machine.Pin.OUT)
nic = wifi_connect()
utime.sleep_ms(1000)
tempTimer = machine.Timer(2)
tempTimer.init(period=DEEP_SLEEP_TIME,
               mode=tempTimer.PERIODIC, callback=manage_temp)
pin.value(1)
````

loboris, 2018-02-16:
Sorry, I had no time to check it. I'll do it over the weekend.

hargathor, 2018-02-17:
After further testing it seems to be related to the mqtt topic. This is an overloaded topic and one of the publisher on it seems to be stuck on a loop (from time to time) and delivering tons of messages. This is what caused the corrupt heap. 
After going on another topic (an fixing this loop bug) the issue didn't occur for a 24 hours test.

kevinkk525, 2018-02-17:
That's interesting. maybe you can make a reproducible small code with example topics that trigger it.
Your posted code doesn't contain any publishing of topics though.. Therefore the bug must be somewhere else?

hargathor, 2018-02-17:
Yes I can try to come up with a script publishing the same kind of messages to try to reproduce the issue. 
The bug was in one my script on my domoticz platform, rookie mistake: I was stuck in a while loop if a message was formatted a certain way. In this loop I was publishing the state of a sensor

kevinkk525, 2018-02-17:
Ok so it was a bug in your script not the mqtt library?
Then we don't need an example but good that you fixed it.

hargathor, 2018-02-17:
Actually I think that if you subscribe to a overloaded topic you can trigger the same corrupted heap.

loboris, 2018-02-17:
The system should not crash, no matter what the content of the script is. If something goes wrong the exception should be raised.
I'll try to reproduce this and find what part of the code is responsible for the crash.

BluSpanner, 2018-02-18:
A bit of testing with MQTT using the following...
- sample DS18B20
- pub to MQTT
- deepsleep 60s
- loop.

MQTT publish looks to have an issue.
Note: The panic is re-producible - but not every time through the loop

```sh
MQTT PUB topic=sensors/30AEA44543D0/sw/update_interval_seconds  message=60.0
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x00008100  PS      : 0x00060a30  A0      : 0x8012c7f8  A1      : 0x3ffb7c50
A2      : 0x3ffca1c0  A3      : 0x00000001  A4      : 0x00000008  A5      : 0x3ffb7d00
A6      : 0x3ffb7c98  A7      : 0x3ffb7d00  A8      : 0x801225d7  A9      : 0x3ffb7c30
A10     : 0x3ffca1c0  A11     : 0x00000001  A12     : 0x00000008  A13     : 0x3f418f6c
A14     : 0xfffffffe  A15     : 0x00000001  SAR     : 0x0000000c  EXCCAUSE: 0x00000014
EXCVADDR: 0x00008100  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x00008100:0x3ffb7c50 0x4012c7f5:0x3ffb7c70 0x40124a7e:0x3ffb7d10 0x40120695:0x3ffb7d80 0x401206d9:0x3ffb7da0 0x401216db:0x3ffb7dd0 0x401217b5:0x3ffb7e50 0x4012d7cc:0x3ffb7e80 0x40124a7e:0x3ffb7f20 0x40120695:0x3ffb7fa0 0x4012d2df:0x3ffb7fc0 0x40124a7e:0x3ffb8060 0x40120695:0x3ffb8090 0x401206c2:0x3ffb80b0 0x40113ded:0x3ffb80d0 0x40114075:0x3ffb8180 0x4010ea36:0x3ffb81a0
0x4012c7f5: mp_execute_bytecode at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:400

0x40124a7e: fun_bc_call at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x40120695: mp_call_function_n_kw at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x401206d9: mp_call_function_1 at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x401216db: mp_call_function_1_protected at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime_utils.c:33

0x401217b5: mp_handle_pending_tail at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/scheduler.c:57

0x4012d7cc: mp_execute_bytecode at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:1297

0x40124a7e: fun_bc_call at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x40120695: mp_call_function_n_kw at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x4012d2df: mp_execute_bytecode at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:946

0x40124a7e: fun_bc_call at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x40120695: mp_call_function_n_kw at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x401206c2: mp_call_function_0 at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x40113ded: parse_compile_execute at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/utils/pyexec.c:531

0x40114075: pyexec_file at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/utils/pyexec.c:502

0x4010ea36: mp_task at /home/david/esp32/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/main.c:150

CPU halted.
```

loboris, 2018-03-06:
A new method of passing parameters to callback functions is now implemented and those crashes should ne longer occure.
Please reopen if the issue is still present.

closed 2018-03-06 by loboris