loboris/MicroPython_ESP32_psRAM_LoBo #71: Time and Threads

GregCle, 2018-02-14:
Hi,
I noticed a problem with time.sleep_ms () and threads.

- If we use time.sleep_ms (50), the code works correctly.
- If we use time.sleep_ms (5), each thread is used intercalated

I put you a test code to check:

```
import time
import _thread

p33 = Pin(33)
Led_Blue = PWM(p33)
Led_Blue.freq(200)
Led_Blue.duty(0)

def Count_1():
    x = 0
    while True:
        x = x + 1
        print(x)
        time.sleep_ms(5)
        #time.sleep_ms(50)

def Count_2():
    y = 0
    while True:
        y = y + 1
        print('------',y)
        time.sleep_ms(5)
        #time.sleep_ms(50)

_thread.start_new_thread("Count_1",Count_1,())
_thread.start_new_thread("Count_2",Count_2,())

```

Output with time.sleep_ms(5):

### 

```

605
606
607
608
------ 577
------ 578
------ 579
------ 580
------ 581
------ 582
------ 583
... (remove output)
------ 629
------ 630
------ 631
------ 632
------ 633
------ 634
------ 635
------ 636
------ 637
------ 638
------ 639
------ 640
609
610
611
612
613
614
615
616
617
618
... (remove output)
664
665
666
667
668
669
670
671
672
------ 641
------ 642
------ 643
------ 644
------ 645
------ 646
------ 647
------ 648
```

Output with time.sleep_ms(50):

```
------ 4
4
5
------ 5
6
------ 6
7
------ 7
------ 8
8
------ 9
9
10
------ 10
11
------ 11
12
------ 12
------ 13
13
------ 14
14
------ 15
```

Do you have any idea why this is happening?

I use:

FreeRTOS running on CORES BOTH, MicroPython task started on Core App.
Running from OTA_0 partition starting at 0x10000, [MicroPython_1].

 Reset reason: RTC WDT reset
    uPY stack: 31744 bytes
     uPY heap: 96000/5856/90144 bytes

MicroPython ESP32_LoBo_v3.1.18 - 2017-02-07 on ESP32 board with ESP32

Thanks you for your help

kevinkk525, 2018-02-14:
I can't see anything wrong here. maybe you have wrong expectations. A call to time.sleep_ms() does not return the control from the thread. Therefore the loop in thread 1 (with x) is called multiple times until the next thread is being executed. The thread library is preemptive, running one thread for a certain period and then pausing the execution of that thread and continuing the next thread.

If you write that thread functions as asynchronous functions with asyncio you would get the expected result:
---4
4
---5
5
---6
6
...

GregCle, 2018-02-14:
Hi,

I misspoke.

With version 2 of the firmware, no matter the time of the time.sleep_ms () I always output (Output 5 ms or 50 ms):
```
------ 4
4
5
------ 5
6
------ 6
7
------ 7
------ 8
8
------ 9
9
10
------ 10
11
------ 11
12
------ 12
------ 13
```

This is normal

But, what I do not find normal, it is in the version 3 of the firmware, it is the breaks are produced with time.sleep_ms (5) which are not produced in version 2 of the firmware.

The breaks with time.sleep_ms (5) in firmware 3 are around 200-300 ms.

According to my test code (time.sleep_ms (5)), Count_1 runs and Count_2 does not do anything for 200-300ms. Then Count_2 runs and Count_1 does not do anything for 200-300ms.

Normally, Count_1 and Count_2 should not have a break so long and should have an output described above.

I hope I was clearer.

Thank you

kevinkk525, 2018-02-14:
That I understand. The time one thread is running should be less than 200-300ms.

loboris, 2018-02-16:
The `time.sleep_ms ()` blocks other threads for sleep times less than 10ms, that is the probable cause for this behaviour.
For sleep times >10ms the thread is sleeping during the sleep time and other threads can run.
When there is no sleep time in the thread, threads are switched by MicroPython VM scheduler every 8 instructions. It can be changed by setting the `MICROPY_SCHEDULER_DEPTH` in `esp32/mpconfigport.h`

closed 2018-03-02 by GregCle