loboris/MicroPython_ESP32_psRAM_LoBo #72: Expose SPI from TFT driver

tuupola, 2018-02-16:
Currently I have not found a way to access the SPI used by the display driver. Creating a new SPI instance with same settings almost works. It however garbles the display output. Is it possible to expose the SPI instance from the driver, so it would be possible to access it with something like:

```python
tft = display.TFT()
tft.init( ... )
tft.spi.write(buf)
```

Or even better use composition while creating the TFT instance. This way both TFT driver and user code would have access to the SPI.

```python
spi = SPI(
    miso=Pin(19), 
    mosi=Pin(23, Pin.OUT), 
    sck=Pin(18, Pin.OUT), 
    spihost=SPI.VSPI, 
    baudrate=2600000
)

tft = display.TFT()
tft.init(
    tft.ILI9341,
    width=320,
    height=240,
    spi=spi
    cs=TFT_CS_PIN,
    dc=TFT_DC_PIN,
    rst_pin=TFT_RST_PIN,
    ...
)
```

loboris, 2018-02-16:
I'm working on some changes to the SPI and Display modules and the spi driver at the moment, to address some problems.

For example, there was a serious problem when using the spi display and the SDCard in SPI mode at the same time caused by the way the esp-idf's sdcard spi-mode driver is implemented.
I've fixed it now (by patching the esp-idf `sdspi_host.c`) and it works good.
That solves the problem with using the display and sdcard at the same time on M5Stack, for example.

It will be possible to initialize the TFT display by passing the **spi object** argument.
The old method (passing mosi,miso,sck) will still be available, it will create the spi object internally and there will be a method to pass it to the user (`tft.getspi()`).

I have to do some more testing, it will probably be committed this weekend or early next week.

loboris, 2018-02-20:
@tuupola

I have finished the rewritting the display module. It now uses the SPI object internally and it could be exposed to the user.

I was thinking about possible uses of the exposed spi instance, but I'm not shure I can find one.

Could you, please, give some example of using the exposed display spi instance from the user program?
Maybe it would be more usefull to expose some lower level functions, like write_address, write_command/read_command, write_data/read data ?

tuupola, 2018-02-20:
One could for example write specific hardware commands to the display driver. Your suggestion to expoose `write_command()` would fit my use case too.

loboris, 2018-03-06:
Some low level display commands are now added, I'll close the issue, please reopen if needed.

closed 2018-03-06 by loboris

zouden, 2019-02-09:
Which commands were added exactly? I would like to initialise the TFT using an SPI object but the wiki doesn't show how to do that.