loboris/MicroPython_ESP32_psRAM_LoBo #73: How to flip ssd1306 oled display

ppontre, 2018-02-18:
Hello Boris, 
I'm trying to set up Lobo upy on an ESP32 TTGO board, with ssd1306 and battery.
I succeeded in installing LoBo ypi (great work!).....
I can display text to the ssd1306 display, but i can't sort out how to flip upside down the display content.
Is there some easy 1-command way to flip it ?
Thanks in advance for yr help
Paolo

loboris, 2018-02-21:
ssd1306 display uses the MicroPython framebuffer. I don't think there is a function to flip the buffer at the moment.
I'll check if flipping/rotating the framebuffer could be implemented.

ppontre, 2018-02-21:
Thanks a lot, Boris,
I found this command and it worked fine (the only problem is that const command in SSD1306 don’t work, but it’s not a great problem.
Here’s the snippet of code:

from micropython import const
import machine
from ssd1306 import SSD1306_I2C
import time
WIDTH = const(128)
HEIGHT = const (64)
sda_pin = machine.Pin(5)
scl_pin = machine.Pin(4)
i2c = machine.I2C(scl=scl_pin, sda=sda_pin, speed=400000)
ssd = SSD1306_I2C(WIDTH, HEIGHT, i2c)
import freesans20
from writer import Writer
wri2 = Writer(ssd, freesans20, verbose=True)
ssd.write_cmd(0xa0)
ssd.write_cmd(0xc0)
Writer.set_clip(True, True)
Writer.set_textpos(0, 0)
#wri2.printstring('Hello\n')
#wri2.printstring('World')
ssd.show()

Inviato da Posta<https://go.microsoft.com/fwlink/?LinkId=550986> per Windows 10

ppontre, 2018-02-21:
Hi Boris,
Any idea about bluetooth (standard & Ble) support in uPy ?
Thx & Regards
Paolo
-------------------------------------------------------------------------------------------------------

closed 2018-03-06 by loboris