loboris/MicroPython_ESP32_psRAM_LoBo #74: gsm.atcmd response parameter issue

RoundBobin, 2018-02-19:
I'm using the latest ESP32 firmware build from Wed Feb 7 17:51:08 2018 +0100.

Trying to get SIM card balance with the help using USSD request.
For this purpose, I'd like to get a full AT command output response, including the USSD message body.
Delimiter '15\r\n' at the end of the message looks fine as an end of message marker.
But it seems that response parameter inside gsm.start does not work correctly.

Please see an example below.

Type 'help()' (without the quotes) then press ENTER.

>>> import gsm
>>> import time
>>> gsm.start(rx=27, tx=26, baudrate=115200, apn="internet", user='', password='')
True
>>> while not gsm.status()[0] == 89: time.sleep(1)
... 
>>> gsm.atcmd('AT+CUSD=1,\"*111#\"', timeout=20000, response='15\r\n')
**'\r\nOK\r\n'**
>>> 

Thus function does not wait for the response and returns '\r\nOK\r\n'

loboris, 2018-02-21:
Sorry, I haven't the time to check it yet.
I'll report back when I'll have more information about the issue.

L4grange, 2019-04-15:
Hello! Do you have any updates on the situation? I'm trying to send a "sleep" command to the gsm module ("AT+CSCLK=1") but I only get an empty response '' when using the gsm.atcmd() function. It also returns immediately and does not wait for the response.

AndriyPytel, 2021-03-14:
Same issue,
`>>> gsm.atcmd('AT+CUSD=1,"*111#"', timeout=1000, printable=True)`

```
I (786978) [PPPOS CLIENT]: AT COMMAND: [AT+CUSD=1,"*111#"..]
I (787151) [PPPOS CLIENT]: AT RESPONSE (to buffer): len=6
'..OK..'
```
If AT response have go to buffer, so how to open get data from this buffer?

TravisDuck999, 2021-05-06:
Hi, I know this is old but I would love to know how to fix this as well, any ideas ?

AndriyPytel, 2021-05-17:
I had use it by uart connectin. But you must disconnect gsm in your code, becouse uart in that moment not availuble.
And you more trick. It's hapens becouse, AT commands. returns in few lines with delay. So gsm.atcmd gets the first one and gives it to you. If you need only AT commands better to use only uart, in another case use swiching between gsm and uart