loboris/MicroPython_ESP32_psRAM_LoBo #75: Wiki MQTT TLS

bdespatis, 2018-02-19:
Hi,

I would like to know how I can point a CA certificate into MQTT Module so I can use TLS (secured connection). I can see user/password, but I don't see a SSL Argument or Certificate Argument.

Thanks for sharing

Benoit

loboris, 2018-02-21:
You can use secure connection, SSL is used if you set `secure=True` argument when creating the mqtt instance object.
The certificate is not checked. Look at this [discution](https://www.esp32.com/viewtopic.php?f=13&t=2163&p=10317&hilit=mqtt+certificate#p10249)
I'll see if it is possible to enable checking the certificate.

docwelch, 2018-04-03:
@loboris - I notice that the esp32_mqtt library has moved to espmqtt in the last couple of months. It appears that there is support for checking the certificate in the new library.  Is moving to the new library a possibility?

loboris, 2018-04-03:
Yes, it will be updated soon.

bdespatis, 2018-04-19:
Hi Boris,

Do we have an estimate of when we can use ssl param inside mqtt module ? Thanks ! It is needed to connect to AWS. We could use simple.py (MQTT Client under MicroPython) if we can't wait.

Regards,
Benoit

Josverl, 2018-04-26:
SSL works for me , i've not tested it against AWS though and I have used the code including the changes to remove unwanted lwt settings. See #126 
id suggest to begin checking if AWS IOT does or does not support LWT and the hardcoded LWT path that is in the current firmware.

loboris, 2018-05-13:
The **mqtt** module was updated, the new mqtt library is now used which should solve the **lwt** issue.
The [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/mqtt) is updated and includes an example to connect to ThingSpeak (tested and works).
Importing the CA certificate from file is now supported.

closed 2018-05-13 by loboris