loboris/MicroPython_ESP32_psRAM_LoBo #76: available memory decrease with mp_obj_dict_t inside a module

gdallaire, 2018-02-21:
Hello,

I try to implement an iBeacon scanner within a C module accessible from micropython. I took the demo example from esp-idf: "iBeacon scanner". 

When a beacon is detected, a callback is called to pass the information to python using "mp_sched_schedule(). The data (UUID, RSSI, etc) is encapsulated in a mp_obj_dict, here is a section of this code :

```
....
case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
...
...
              mp_obj_dict_t *dct = mp_obj_new_dict(0);
              mp_obj_dict_store(dct, mp_obj_new_str("UUID", 4, false), mp_obj_new_bytes(ibeacon_data->ibeacon_vendor.proximity_uuid, 16));
                mp_obj_dict_store(dct, mp_obj_new_str("RSSI", 4, false), mp_obj_new_int(ibeacon_data->ibeacon_vendor.measured_power));
                mp_sched_schedule(scan_callback, dct);
            }
            break;
...
```
The issue is that memory usage increases each time the callback is called.

After running less than some minutes without calling gc.collect(), the system crashes (see the LOG output tracing available heap) :

```
...
...
LOG: gc.free_mem() == 25888
LOG: gc.free_mem() == 25552
LOG: gc.free_mem() == 25216

Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x40131267  PS      : 0x00060e30  A0      : 0x80131253  A1      : 0x3ffb3230  
A2      : 0x3ffb3230  A3      : 0x3ffb3230  A4      : 0x3ffd0724  A5      : 0x00000003  
A6      : 0x3ffd0904  A7      : 0x00000000  A8      : 0x80131267  A9      : 0x3ffb3210  
A10     : 0x00000000  A11     : 0x3ffd0804  A12     : 0x00000057  A13     : 0x3ffd080c  
A14     : 0x0000000a  A15     : 0x3ffbcb70  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x40116274  LEND    : 0x401162d2  LCOUNT  : 0x00000000  

Backtrace: 0x40131267:0x3ffb3230 0x40131250:0x3ffb3260 0x40131250:0x3ffb3290 0x40131250:0x3ffb32c0 0x40131250:0x3ffb32f0 0x40131250:0x3ffb3320 0x40131250:0x3ffb3350 0x40131250:0x3ffb3380 0x40131250:0x3ffb33b0 0x4013128d:0x3ffb33e0 0x401167f5:0x3ffb3400 0x40116169:0x3ffb3430 0x40124b67:0x3ffb3450 0x40125499:0x3ffb3470 0x40135e25:0x3ffb3490 0x401630b1:0x3ffb34b0 0x40160e06:0x3ffb34e0

CPU halted.

```
If I insert a periodic timer to call gc.collect() as soon as gc.free_mem() reports a certain level of low memory, I can get the ESP32 to survive some hours. But eventually I always get also the same kind of fatal error : 

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
....
...
```
I tried to run the same code but without creating and passing the mp_obj_new_dict object to the  callback. Result: no problem with heap usage.

If the mp_obj_new_dict object is created but not passed to the callback, the heap usage increase.

I guess something goes wrong when dealing with a mp_obj_new_dict(0) object created inside this function. 

Any ideas or suggestions ?

loboris, 2018-02-21:
Your dictionarry created in `case ESP_GAP_BLE_SCAN_RESULT_EVT:` is allocated on MicroPython heap and will stay there until not used anywhere.
I suppose you have defined MicroPython function (something like `def scan_func(res):`) assigned to `scan_callback` which receives the dictionary argument.
In this function you should free the dictionary argument: `res = None` or `del res`, that way the garbage collector will be able to free the memory used by it.

gdallaire, 2018-02-23:
Is there a way to explicitly free an object created with mp_obj_new_dict from the C module ?

loboris, 2018-02-23:
I have analyzed this issue more carefully, and here are my findings:

A dictionary object is created and populated using
```
mp_obj_dict_t *dct = mp_obj_new_dict(0);
mp_obj_dict_store(dct, mp_obj_new_str("UUID", 4, false), mp_obj_new_bytes(ibeacon_data-ibeacon_vendor.proximity_uuid, 16));
mp_obj_dict_store(dct, mp_obj_new_str("RSSI", 4, false), mp_obj_new_int(ibeacon_data-ibeacon_vendor.measured_power));
```
When the heap space is allocated for the new string object it is ultimately done with `gc_alloc()` function from _py/gc.c_

If the **gc threshold** is reached, `gc_collect()` is called from inside this function, which causes the **crash** because it is called **outside** the MicroPython environment.

The **solution** is to temporary disable **gc** when creating the dictionary (or any other) object from code which is not part of the MicroPython environment (other FreeRTOS tasks, event callbacks, etc):
```
int gc_auto = MP_STATE_MEM(gc_auto_collect_enabled);
MP_STATE_MEM(gc_auto_collect_enabled) = 0;
mp_obj_dict_t *dct = mp_obj_new_dict(0);
mp_obj_dict_store(dct, mp_obj_new_str("UUID", 4, false), mp_obj_new_bytes(ibeacon_data-ibeacon_vendor.proximity_uuid, 16));
mp_obj_dict_store(dct, mp_obj_new_str("RSSI", 4, false), mp_obj_new_int(ibeacon_data-ibeacon_vendor.measured_power));
mp_sched_schedule(scan_callback, dct);
MP_STATE_MEM(gc_auto_collect_enabled) = gc_auto;
```
- Current auto state is saved: `gc_auto = MP_STATE_MEM(gc_auto_collect_enabled)`.
- Auto collect is disabled: `MP_STATE_MEM(gc_auto_collect_enabled) = 0`
- Object allocation is performed.
- Auto collect state is restored: `MP_STATE_MEM(gc_auto_collect_enabled) = gc_auto`

I've tested this approach and it looks it works.
I've run the code similar to yours which was crashing after some time. After the change, the code was running for hours with no problem.
It may be interesting to monitor the heap usage with `micropython.mem_info(1)`.

There is no need to free the dictionary object, it will be garbage collected during the normal MicroPython run.

loboris, 2018-03-06:
A new method is now implemented for passing the parameters to the callback scheduler from the functions running in tasks, interrupts or event handlers.
Those kind of errors should no longer occur.

closed 2018-03-06 by loboris